PX4/PX4-user_guide #54: Doc Bug: Full Parameter Reference · PX4 User Guide  · FW_PSP_OFF Description

jeffstreet, 2017-08-18:
FW_PSP_OFF description currently says:
> An airframe specific offset of the pitch setpoint in degrees, the value is added to the pitch setpoint and should correspond to the typical cruise speed of the airframe.

The word "speed" might be replaced with, "... typical cruise *pitch angle (angle of attack)* of the airframe."

A similar change could be made to the description of FW_RSP_OFF by replacing "speed" with "roll angle".

Bug Page: [Full Parameter Reference · PX4 User Guide](https://docs.px4.io/en/advanced_config/parameter_reference.html)

hamishwillee, 2017-08-18:
Hi Jeff, can you please repost in PX4/Firmware repo? This information is generated from markup in source code.

hamishwillee, 2017-11-10:
This has been cross posted to the right place.

closed 2017-11-10 by hamishwillee