PX4/PX4-user_guide #75: Multicopter PID Tuning Guide review

nicolaerosia, 2017-10-11:
Hello,

I'm trying to understand the [Multicopter PID Tuning Guide](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html) and I have some questions:

**"Set all MC_XXX_P to zero (ROLL, PITCH, YAW)"** if these parameters do what they advertise, I don't understand how this will work since they will output 0.
_"The gains actually have an intuitive meaning, e.g.: if the MC_ROLL_P gain is 6.0, the copter will try to compensate 0.5 radian offset in attitude (~30 degrees) with 6 times the angular speed, i.e. 3 radians/s or ~170 degrees/s."_ working this example and replacing 6 with 0, it will output 0 degrees/s.

**"Keep the multi rotor in your hand and increase the thrust to about 50%, so that the weight is virtually zero. "**
In which mode? Manual Stabilized?

**I Gain Tuning
If the roll and pitch rates never reach the setpoint but have an offset, add MC_ROLLRATE_I and MC_PITCHRATE_I gains, starting at 5-10% of the MC_ROLLRATE_P gain value.**
How can I see the roll and pitch rates? what about setpoint? Are we still in the situation where I keep the drone in my hand, titled, with 50% throttle ?

@hamishwillee @dagar

Thanks,
Nicolae

hamishwillee, 2017-10-13:
@nicolaerosia 

>> "Set all MC_XXX_P to zero (ROLL, PITCH, YAW)" if these parameters do what they advertise, 

> I don't understand how this will work since they will output 0.

This is a *preparation step* ("Step 1: Preparation") - so essentially you're setting everything to a nothing state before you start tuning each of the individual values. So yes, you're right, but it is intentional.

The rest of the questions seem reasonably and I certainly can't answer. While I hope @dagar will answer them, can you please raise them here: http://discuss.px4.io/c/multicopter
Then we can update the docs as we get answers.

dagar, 2017-10-13:
> "Keep the multi rotor in your hand and increase the thrust to about 50%, so that the weight is virtually zero. "
> In which mode? Manual Stabilized?
Yes manual/stabilized

**How can I see the roll and pitch rates? what about setpoint? Are we still in the situation where I keep the drone in my hand, titled, with 50% throttle ?**
In flight you can use QGC analyze and plot ATTITUDE (http://mavlink.org/messages/common#ATTITUDE) and ATTITUDE_TARGET (http://mavlink.org/messages/common#ATTITUDE_TARGET).

Offline with a .ulg see vehicle_rates_setpoint and vehicle_attitude.

hamishwillee, 2017-10-16:
@dagar As above "....Are we still in the situation where I keep the drone in my hand, titled, with 50% throttle ?"

Or to put it another way the only instruction in the doc about where you have the vehicle and how much you have it powered is that one above. Does that mean that ALL tuning is done with that situation. IF not, which sections do you do something else in, and what is it you do?

hamishwillee, 2017-10-16:
So I've partially fixed this in #78. @nicolaerosia and @dagar please check it is improved (not completely done!). Still waiting on answers to above.

hamishwillee, 2017-10-18:
@dagar Ping on https://github.com/PX4/px4_user_guide/issues/75#issuecomment-336756696

Is all testing done with  drone in hand, tilted, with 50% throttle ?" If not, in what steps is this changed, and how?

dagar, 2017-11-07:
I'm not sure what the best answer is, but no, after that I'd actually start hovering.

hamishwillee, 2017-11-07:
OK, so following your response and re-reading this, it sounds like all the tuning except [feed forward tuning](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html#feed-forward-tuning) kind of needs to be done in the hand so that you can change orientation as indicated. Once tuning complete the rest of testing can be done in hover.

So I will leave this as is. Issue closed :-)

closed 2017-11-07 by hamishwillee