PX4/PX4-user_guide #96: Add takeoff information to altitude and position control mode docs

hamishwillee, 2017-11-15:
For altitude and position mode, multicopter, add not about what counts as "takeoff"/how to takeoff.

The line to add is something along the lines of:

> The vehicle will take off if the stick is raised above 62.5% percent.

Where  62.5%  refers to the full range from the throttle stick bottom position. So in effect, to just above the large dead zone in the middle?

hamishwillee, 2017-11-15:
@dagar The statement "The vehicle will take off if the stick is raised above 62.5% percent." is true (it is hard coded) but what does this % refer to?

I THINK that it refers to the full range from the throttle stick bottom position. So in effect, to just above the large dead zone in the middle. Would that be correct?

dagar, 2017-11-17:
Yes it's the full range starting from the bottom. It makes more sense when the throttle stick is spring loaded and centers itself.

bys1123, 2019-02-05:
@hamishwillee I think this should add to doc when you have time.

hamishwillee, 2019-02-05:
Done in #433

bys1123, 2019-02-06:
Wow, so fast, thanks!

PR #433 (2019-02-15): Add info about takeoff in MC position and altitude modes
Fixes #96
closed 2019-02-15 by hamishwillee