PX4/PX4-user_guide #100: Fix up precision land flight mode parameters

hamishwillee, 2017-11-20:
[Precision land]() was merged in https://github.com/PX4/px4_user_guide/pull/98/files and updated in https://github.com/PX4/px4_user_guide/pull/99

@ndepal None of the parameters mentioned appear in the [parameter reference](https://docs.px4.io/en/advanced_config/parameter_reference.html) - e.g. LTEST_MODE, LTEST_SCALE_X, LTEST_SCALE_Y etc.

Can we get them added. Not only does this force proper documentation, but it means that all the links I just added to the reference will actually work :-0

ndepal, 2017-11-20:
I'm not sure what is required for the parameters to show up there.
I'm guessing one criterion is that the code needs to be in master, which it is no yet.
If the param reference is not generated from the `prarams.c`, please let me know what needs to be added.

hamishwillee, 2017-11-20:
My fault - this should not be merged until in master. At that point the parameters if present should get picked up. I vote we look at this again once this is merged. IN the meantime I added a note on the topic so people won't try and use it yet!

hamishwillee, 2025-01-29:
Closing. I believe this was all addressed.

closed 2025-01-29 by hamishwillee