PX4/PX4-user_guide #122: Need a page that explains how to find a parameter

LorenzMeier, 2017-12-30:
@hamishwillee Here is the image for it. Just need a central page for this and general settings. Then we can reference statements like "set parameter ABC to 1.0" to that page.

[image]

hamishwillee, 2018-01-03:
The [parameters doc](https://docs.qgroundcontrol.com/en/SetupView/Parameters.html) in QGC should be usable for this purpose but is a bit out of date. I've updated it (https://github.com/mavlink/qgc-user-guide/pull/125) and I'll cross post with minor modification into the user guide.

hamishwillee, 2018-01-03:
Addressed by #126

closed 2023-05-24 by hamishwillee