PX4/PX4-user_guide #128: Missing information in setup and sensor calibration procedure

priseborough, 2018-01-08:
Documentation for the setup and sensor calibration procedure is not providing the following information:

- Magnetometer calibration should be done with usb disconnected and the vehicle in a flight configuration and failure to do this can result in a  "mag sensors inconsistent" error. This is a particular problem with the pixracer where USB connection can produce large levels of magnetic interference.
- Raw comparison data for all sensors can be logged by setting SDLOG_MODE = 1 and SDLOG_PROFILE = 64
- Sensors with poor performance can be disabled using the CAL_..._EN parameter

Providing this information would help prevent user issues like this one https://github.com/PX4/Firmware/issues/8098

hamishwillee, 2018-01-09:
> Magnetometer calibration should be done with usb disconnected and the vehicle in a flight configuration and failure to do this can result in a "mag sensors inconsistent" error. This is a particular problem with the pixracer where USB connection can produce large levels of magnetic interference.

Please confirm:
1.  the user would have to have telemetry radio of some kind as the only way to connect to QGC to perform the calibration? That sounds like a bit of a limitation.
2. When you say "vehicle in a flight configuration" what do you mean? What other configuration can it be in?

For this one proposal is to update https://docs.px4.io/en/config/compass.html putting connection requirements in point "1"

hamishwillee, 2018-01-09:
> Sensors with poor performance can be disabled using the CAL_..._EN parameter

When/what sensors would you do this for? I assume this is primarily for the case of magnetometers where you can have poorly behaving internal compasses that you can't remove so you want to disable them? (presumably you could just remove a badly behaving external compass).

What other sensors might this apply to? 

If this is just general advice, perhaps we should create some sort of calibration troubleshooting/debugging topic in the user guide?

hamishwillee, 2018-01-09:
> Raw comparison data for all sensors can be logged by setting SDLOG_MODE = 1 and SDLOG_PROFILE = 64 ([bit 6 set](https://docs.px4.io/en/advanced_config/parameter_reference.html#SDLOG_PROFILE)

Sounds useful.

1. Is there any information on how the comparison data might be used/interpreted?
2. I don't think this should live inside the basic configuration topics/section, but could perhaps be in a troubleshooting section and linked from them. Maybe we should extend https://dev.px4.io/en/debug/sensor_uorb_topic_debugging.html in the devguide to be more generic (i.e. cover this option to?)

hamishwillee, 2023-05-24:
The third point has been addressed:
- Sensors with poor performance can be disabled using the CAL_..._EN parameter

The other two are:

- Magnetometer calibration should be done with usb disconnected and the vehicle in a flight configuration and failure to do this can result in a "mag sensors inconsistent" error. This is a particular problem with the pixracer where USB connection can produce large levels of magnetic interference.
- Raw comparison data for all sensors can be logged by setting SDLOG_MODE = 1 and SDLOG_PROFILE = 64

@bresch Would this still be relevant? I.e. do we need to worry about USB? If so, how do you do the calibration if you don't have a telemetry radio?

bresch, 2023-05-26:
Yes, I think it is still relevant. For the USB, yes, this is an issue with the Pixracer but it might apply to other boards as well (maybe future ones too).

hamishwillee, 2023-06-07:
Thanks @bresch . Fixed by https://github.com/PX4/PX4-user_guide/pull/2563

PR #2563 (2023-06-07): Magnetometer calibration - disconnect USB
Fixes #128

USB can cause magnetic interference. This adds a note for Pixracer, which is particularly affected. Also adds removing USB to compass calibration as a precondition "if at all possible", and information on how to get logs.
closed 2023-06-07 by hamishwillee