PX4/PX4-user_guide #146: PX4 Basic Concepts > Headings and Directions > Multicopter diagram

hamishwillee, 2018-02-13:
[PX4 Basic Concepts > Headings and Directions](https://docs.px4.io/en/getting_started/px4_basic_concepts.html#heading-and-directions)

Local reference frame in the quadcopter image is wrong. According to the right hand rule, the light blue y-axis should point in the ‘east’ direction if we are following NED convention.

[image]

@barzanisar Assigned to you for now!

barzanisar, 2018-02-19:
@hamishwillee It will be helpful to know who added this diagram in the first place and how he/she created it. I can edit it my way but since I am not a drawing/photoshopping expert, my edited version will look ugly with uneven arrow thickness etc.

hamishwillee, 2018-02-19:
@tubeme Did you do the diagrams here: https://docs.px4.io/en/getting_started/px4_basic_concepts.html#heading-and-directions ?

We'd like to make a minor edit, but it would be helpful if you could point us to the drawing package/original versions used?

tubeme, 2018-02-19:
Yes I did them. I could do the edit. What should be edited? I can make them right away.

barzanisar, 2018-02-19:
According to the right hand rule, the light blue y-axis should point in the ‘east’ direction if we are following NED convention.

tubeme, 2018-02-19:
It is pointing to the "east" in this drawing. The problem is that the drawing is looked at the bottom of the quad.

tubeme, 2018-02-20:
@hamishwillee @barzanisar Do you need another drawing for the quad despite this one is correct?

barzanisar, 2018-02-20:
Ahh now I get it. I couldn't understand the view when I first looked at it. Is it possible to make a top/front view of this diagram like you did with plane and helicopter?

tubeme, 2018-02-20:
OK but I have to find a good clip art and correct it. I will post it tomorrow.

hamishwillee, 2018-02-20:
Hi @tubeme . Thanks for the clarification/reasoning.

If you are able to find acceptable clipart in the same orientation as the other images that would be better/less prone to confusion. If not, the image is actually correct and we can live with it 

Can you please also provide a link to the source of the images you use here so that others can edit in future if needed.

tubeme, 2018-02-21:
Here you go.
[image]

[image]

And rar with the images
[Heding.zip](https://github.com/PX4/px4_user_guide/files/1743997/Heding.zip)

hamishwillee, 2018-02-21:
Thanks @tubeme ! Updated.

closed 2018-02-21 by hamishwillee

tubeme, 2018-02-21:
@hamishwillee I made some little corrections in the first image as well.

hamishwillee, 2018-02-22:
@tubeme Sorry, didn't see the changes. I've added now in https://github.com/PX4/px4_user_guide/pull/162

tubeme, 2018-02-22:
Hey @hamishwillee can you see this page?
http://px4.io/wp-admin/post.php?post=1738&action=edit

This is a draft of a Basic flight modes page from the old users manual. I did not publish it back then because there was constantly changing behaviour in the modes at that time. But anyway I updated it and with some corrections it could fit the current users manual. It has a very nice clip art visualising the modes.

It is very good idea to fit this page in the Getting started section. Because it is silly to show a Flight modes Setup if you did not provide information about the Flight Modes in advance.

Here is a DOC file if you cannot open the link.
[Basic PX4 flight modes.docx](https://github.com/PX4/px4_user_guide/files/1747764/Basic.PX4.flight.modes.docx)

tubeme, 2018-02-22:
Here is the preview link.
http://px4.io/?post_type=docit&p=1738&preview=true

tubeme, 2018-02-22:
My intent was to have a very simple words explanation about modes. For people that do not have a lot of knowledge ie the novice.

tubeme, 2018-02-22:
@hamishwillee I made some corrections in the DOC
[Basic PX4 flight modes II.docx](https://github.com/PX4/px4_user_guide/files/1747833/Basic.PX4.flight.modes.II.docx)

hamishwillee, 2018-02-22:
@tubeme I have created issue for this and assigned it to you: https://github.com/PX4/px4_user_guide/issues/163 :-)

The short response is that I am aware of this doc and have been mining it for information as I evolve https://docs.px4.io/en/flight_modes/ and create individual topics like [Takeoff mode](https://docs.px4.io/en/flight_modes/takeoff.html).

The whole getting started needs revisiting, but I agree with you that a simple introduction to flight modes is needed. I also think our https://docs.px4.io/en/flight_modes/ needs to be pulled out of "Flying" to the top level.

commit 09956b71 (2018-02-21): Merge pull request #161 from hamishwillee/mc_headings

Update heading images to fix #146