PX4/PX4-user_guide #151: Multicopter PID Tuning Guide > Motor band limiting - what is lower side

hamishwillee, 2018-02-13:
[Multicopter PID Tuning Guide > Motor band limiting](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html#motor-band--limiting)

Where it says:
"If one of the rotors leaves this safety band, the total thrust of the system is lowered so that the relative percentage that the controller did output can be satisfied. As a result the multi rotor might not climb or lose altitude a bit, but it will never flip over. *The same for lower side, even if commanded roll is large, it will be scaled to not exceed commanded summary thrust and copter will not flip on takeoff at near-zero thrust.*?

What is the lower side? what is summary thrust?

@bkueng Can you suggest someone to help with this.

bkueng, 2018-02-13:
The summary thrust (in Manual/Acro mode, this is directly the throttle stick input), is the commanded total thrust - if that makes sense.
The lower side is towards zero: imagine if you have a roll input on a quad: for example the right 2 motors need to increase thrust, while the left 2 motors need to decrease thrust. This is added/substracted from the the commanded summary thrust, and thus can become <0 (especially if commanded thrust is low). Now there are 2 options:
- the mixer reduces the commanded roll, such that no motor command is <0 (how it's currently implemented. In addition the mimum commanded throttle is 8% (configurable parameter: MPC_MANTHR_MIN ), so that even at minimum throttle, the vehicle can control the roll, but with reduced authority). This is the safer way to do it, because the commanded throttle will never be scaled up.
- the mixer increases the commanded throttle, so that your roll command can be satisfied (see PR below). The benefit is that you have full attitude authority, and thus better flight performance even with low commanded throttle.

Maybe a graph would help here?

Note that https://github.com/PX4/Firmware/pull/7920 changes this behaviour: configurable via parameter, it will allow the summary thrust to be increased, so that the vehicle can keep the attitude even if the commanded thrust is zero (it also discusses potential dangers).

hamishwillee, 2018-02-23:
Thanks @bkueng , I'm still confused, but I think it hinges about one point "what is the lower side" - since we haven't been talking about any "higher side" there is no context for me to guess what that is. The explanation "The lower side is towards zero:" doesn't answr that question 

Looking at this bit by bit:\
> The lower side is towards zero: imagine if you have a roll input on a quad: for example the right 2 motors need to increase thrust, while the left 2 motors need to decrease thrust. 

Are you saying that if we have a quad that is rolling, the lower side is the one that is going down? So in this case the left two motors?

> This is added/subtracted from the commanded summary thrust, and thus can become <0 (especially if commanded thrust is low). 

So the summary thrust is the total thrust to all motors. As we're rolling we reduce the thrust on two motors, reducing the total thrust, and add thrust to the other two, increasing the total/summary thrust. I don't understand how the summary thrust can drop below zero!

Ignoring the summary thrust, I think what you're saying is that the total commanded thrust must be positive, but a particular motor might end up being commanded negative thrust by our controller model?

> Now there are 2 options:
> the mixer reduces the commanded roll, such that no motor command is <0 (how it's currently implemented.

So basically we say we're not going to roll so far, so the summary thrust is lowered, and the resulting individual motors are now such that none get set below zero. 

> In addition the mi mum commanded throttle is 8% (configurable parameter: MPC_MANTHR_MIN ), so that even at minimum throttle, the vehicle can control the roll, but with reduced authority). 

I think here you're just saying that even at low throttle there will be some level of rolling available. But it doesn't sound like this will have much to do with the motor band limiting?

> This is the safer way to do it, because the commanded throttle will never be scaled up.
> the mixer increases the commanded throttle, so that your roll command can be satisfied (see PR below). The [...]

So what I think you're saying here is that the alternative is to increase the commanded throttle so that none of the individual motors will get a negative thrust command. 

----
Unfortunately tidying this won't be a priority for me. Perhaps clean up as part of  https://github.com/PX4/Firmware/pull/7920 ?

bkueng, 2018-02-23:
> Are you saying that if we have a quad that is rolling, the lower side is the one that is going down? So in this case the left two motors?

Yes

> So the summary thrust is the total thrust to all motors. As we're rolling we reduce the thrust on two motors, reducing the total thrust, and add thrust to the other two, increasing the total/summary thrust. I don't understand how the summary thrust can drop below zero!

You can look at the summary thrust as the average. If you increase the left side by a delta thrust and increase the right side by the same delta, your total thrust stays the same. Maybe this image helps: https://www.intechopen.com/source/html/6587/media/image46.png
PWM is summary thurst. And delta PWM is what you add/substract for the roll command. The important case is if PWM - delta PWM becomes < 0.

> but a particular motor might end up being commanded negative thrust by our controller model?

Exactly.

> So basically we say we're not going to roll so far, so the summary thrust is lowered, and the resulting individual motors are now such that none get set below zero.

Yes, almost: reducing the roll, meaning we reduce delta PWM.

> But it doesn't sound like this will have much to do with the motor band limiting?

Probably not, but it's imporant to mention here.

> So what I think you're saying here is that the alternative is to increase the commanded throttle so that none of the individual motors will get a negative thrust command.

Yes, in the above image this would be PWM.

hamishwillee, 2025-01-29:
Closing. I'm pretty sure this has been addressed.

In any case, don't consider any fixes to this as high value when users can try autotuning, and we have had no comments and relatively low hitcount on that doc since it was introduced.

Can be reopened if there are questions in this area.

closed 2025-01-29 by hamishwillee