PX4/PX4-user_guide #152: What are the different SYS_COMPANION modes

hamishwillee, 2018-02-13:
See https://docs.px4.io/en/advanced_config/parameter_reference.html#SYS_COMPANION

There are a bunch of modes - e.g.  OSD, Command Receiver, Companion Link, Crazyflie , ESP8266 ,Iridium 

What are each of these modes for, and how do they differ (ie what do they send)? Is there any missing documentation associated with them?

bkueng, 2018-02-13:
Command Receiver: I think this is historic, from a time when not all mavlink instances allowed to receive commands (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.h#L443). Now they all do.

OSD: defines the mavlink stream (set of activated messages & their rates). All the others are for a specific hardware (e.g. ESP8266 for WiFi, the one that is also used on a PixRacer), and use the optimal settings for their link (it's mostly about what bandwidth is available).
Does that clarify it?

hamishwillee, 2018-02-23:
Thanks very much. The OSD info has been incorporated in a doc. I'm not sure if it is worth following up on Command Receiver (i.e. updating docs to state it is legacy/deprecated?). 

Closing for now.

closed 2018-02-23 by hamishwillee