PX4/PX4-user_guide #163: Getting Started

hamishwillee, 2018-02-22:
Add a lightweight flight modes page here [Getting Started](https://docs.px4.io/en/getting_started/) because it silly to show a Flight modes Setup if you did not provide information about the Flight Modes in advance.

There is a document from @tubeme: [Basic PX4 flight modes II.docx](https://github.com/PX4/px4_user_guide/files/1747833/Basic.PX4.flight.modes.II.docx) (also http://px4.io/?post_type=docit&p=1738&preview=true for those with PX4 admin access)

hamishwillee, 2018-02-22:
@tubeme This is assigned to you, as I have been "reprioritised". Hope that is OK? I will add subediting advice.

tubeme, 2018-02-22:
ОК I just have NO know knowledge where to edit the doc/page in the new users manual. So please give me some pointers.

hamishwillee, 2018-02-23:
OK, there is a guide here: https://dev.px4.io/en/contribute/docs.html

I tend to install the Gitbook toolchain locally - follow the instructions [here](https://toolchain.gitbook.com/setup.html) because that makes it easy to add images etc. 

The source itself is in this repo. English versions of files are stored in folders under the en/tree. Images are usually stored in the /assets tree.  Text is written in "more or less" standard github markdown. Files are added for building by editing /en/SUMMARY.md.

It is remarkably easy once you start playing.

tubeme, 2018-03-16:
@hamishwillee Sorry to ask a stupid question but I could not find a way to change the text color. Like to make a text red. 

Any ideas?

hamishwillee, 2018-03-16:
Yes. Don't. 

If you have to, use a `> **Warning**`

If you really have to (you don't), then you can use HTML for this `<span style="color:red;"></span>`

tubeme, 2018-03-16:
Thanks it is actually one HTML... so I could use all the HTML coding right?

Warning should always be in red...

tubeme, 2018-03-16:
The <span style="color:red;"></span> does not work...

hamishwillee, 2018-03-16:
Sure it does in gitbook. But use the warning note by preference.

tubeme, 2018-03-16:
Strange it does not show in px4_user_guide/en/getting_started/basic_flight_modes.md

tubeme, 2018-03-16:
Probably we need this gitbook plugin installed: gitbook-plugin-style

https://www.npmjs.com/package/gitbook-plugin-style

tubeme, 2018-03-16:
`> **Warning**` does not work either...

hamishwillee, 2018-03-18:
Thanks for posting this. Excellent job for your first post. One request - can you please do changes as a PR rather than as a bunch of commits to master? I treat gitbook just like code - I fork the docs and post a PR for all my changes in one go. That way others can review them and help you improve. 

> Strange it does not show in px4_user_guide/en/getting_started/basic_flight_modes.md

I did a search in there an there are no spans/colours. You can see below that adding color works. **BUT TRY AVOID IT**. 

[image]

The warning shows up fine! 

I'm going to do some tidy of this today. I'll cross link.

hamishwillee, 2018-04-19:
Done (mostly). Thanks @tubeme .

closed 2018-04-19 by hamishwillee