PX4/PX4-user_guide #178: Proposed flight mode docs structure and iconography

hamishwillee, 2018-03-22:
This proposal is that we simplify the structure to make it easier to navigate, and use better iconography to provide needed context when you are within the document.

**Background**

The current structure for [Flight Modes Overview](https://docs.px4.io/en/getting_started/flight_modes.html) sorts flight modes by vehicle, type of mode (e.g. auto, manual).  It is quite hard to read because it is so large. When you are inside the document and reading a particular flight mode it is easy to lose track of the fact that you're in MC or FW. There is also repetition of some of the grouping information (auto, manual, assisted). 

My thought is that while it is useful to know there are different categories of modes and to group the modes by category, it is not necessary to have this grouping marked by a heading in the TOC. 

**Proposal: Improve structure:**
1. List the modes in each category in the category overview like this:
  [image]

2. Remove the headings for manual, assisted, auto modes in the TOC so everything is just sorted by vehicle (but still grouped with others in same category)
   - With only one parent heading, it is much easier to know you are in MC or FW
   - There is no duplication of the information about categories.
  [image]

**Proposal: Improve icons:**

The proposal is to add a bar of icons in each of the modes. This will provide additional context in the same way that the GPS icon currently does (though we should if possible change that because it implies "GPS" and we support other position technologies). 

We could have icons for:
- Requires position
- Requires remote control (manual or assisted) - I don't think we need an "assisted" icon.
- Does not require remote control (Auto) - ie remote control crossed out?
- Vehicle (e.g. MC icon indicates this is MC mode)
- Other ?

For example:
  
[image]

@tubeme @bkueng @barzanisar - What do you think? What icons should we use? 

----

Discussion was [originally here](https://github.com/PX4/px4_user_guide/pull/175#issuecomment-375189471) (closed PR).

hamishwillee, 2018-03-22:
FYI some possible icons: 

----

A few other icons are listed below....

* [image] - [from here](https://www.flaticon.com/free-icon/gps-fixed-indicator_60834)
* [image]  - [from here](https://www.iconfinder.com/icons/126573/gps_location_map_pin_icon#size=32)

* [image] - from [here](https://www.google.com.au/search?q=free+small+icon+gps&dcr=0&tbm=isch&tbo=u&source=univ&sa=X&ved=0ahUKEwiTve2Y-P7ZAhVBjpQKHW4ID7kQ7AkINg&biw=1229&bih=587#imgrc=rlid6HG39D63OM:)

And maybe this one for rc:
[image]
 (from [here](http://www.adelaideadvertising.com/free-drone-icon-design-to-download/)

hamishwillee, 2018-03-23:
@tubeme Very much leaning towards at least doing the restructure proposed above. In the [flight mode table](https://docs.px4.io/en/flight_modes/flight_modes_summary.html#multicopter) you will see that I don't organise by assisted/manual so I can relate each mode to the one before. 

ie restructuring would allow me to reorder the modes and use the "ladder principle" you suggested.

bkueng, 2018-03-23:
I like your proposal - and the set of icons. Maybe add one that shows the level of difficulty?
For the position icon, I would probably go for https://www.flaticon.com/free-icon/gps-fixed-indicator_60834.

hamishwillee, 2018-03-26:
Thanks. OK, so first part done (reorg - not icons): https://github.com/PX4/px4_user_guide/pull/183

tubeme, 2018-03-26:
Looks good to me. I got some icons I noticed that can be used for positioning device as well.
[image]
[image]
[image]

For the difficulty level the icon is too detailed for the small size it appears on the page.

So probably using just a dot with a color will be more iconic and easy to perceive.

We can use green, yellow and red like in the traffic lights?

tubeme, 2018-03-26:
This one is not bad at all and has signature to it.
[image]

hamishwillee, 2018-03-27:
@tubeme I do like that last position image, but it is a "paid for" one. I used a similar icon that was free as shown below. Updated the difficulty icon using number and and colour as you suggested - decided to have the 6 level options. 

[image]

Better options still welcome! If you do find one though, please provide URL.

hamishwillee, 2018-03-27:
Closing this, as it is being addressed in https://github.com/PX4/px4_user_guide/pull/184

closed 2018-03-27 by hamishwillee