PX4/PX4-user_guide #181: Doc Bug: Flight Modes · PX4 User Guide

hamishwillee, 2018-03-23:
[Flight Mode Overview > Manual/Stabilized mode](https://docs.px4.io/en/getting_started/flight_modes.html#manualstabilized-mode) - the image isn't quite right.

1. On right, text "Level Flight / Stabilized" - "Stabilized" is miss-spelled in image.
2. The left stick is labeled rate controlled. But actually only the yaw input is rate controlled, not the throttle. 

IMO for this case having the text label "Angle" and "Rate controlled" isn't useful because it won't mean anything to most users. You and I know that angle implies that the vehicle will move left/right or forward/back at a constant speed for the current throttle, but not a new user.

tubeme, 2018-03-26:
I agree for the Angle and rate control... Could not find a better words to describe it...

Here are the corrected images.
[image]
[image]

hamishwillee, 2018-03-27:
Thanks @tubeme . For the "FW" image, could you please also change the spelling "Glyde" to "Glide"?

hamishwillee, 2018-03-27:
> I agree for the Angle and rate control... Could not find a better words to describe it...

Angle and angle rate control are fine. The problem is that the Up/Down stick is not rate controlled - only the spin (yaw) is. That said, I don't see any easy way to correct it - this looks unbalanced:
[image]

closed 2023-05-24 by hamishwillee