PX4/PX4-user_guide #193: [For improve] Procedure Clarification - Doc Bug: Airspeed · PX4 User Guide

bigbellmercy, 2018-04-10:
Hi, Thanks for User Guide.

The airspeed calibration procedure need more clarification of meaning for novices to do it.
Following is my question that I don't know:

4. Cover the sensor (i.e. with your hand). --> For what? 'to protect wind from the sensor'?
6. Blow across the sensor. --> By mouth, ok? or some blowing device is needed?
And blow timing is not clear if we use mouth to blow, how many times at when blow?

Bug Page: [Airspeed · PX4 User Guide](https://docs.px4.io/en/config/airspeed.html)

bigbellmercy, 2018-04-10:
The other [post](http://discuss.px4.io/t/airspeed-sensor-calibration/5083) requests the same request as mine.

After I've succeeded in the calibration, I noticed the problem the document:

The key point of my mistake was that I was blowing air by mouth perpendicular to the pitot tube because of the sentence 'Blow **across** the sensor'.
After I blow air **parallel** to the pitot tube **by mouth shortly** in front of the tip of pitot tube, calibration did not failed.
Other people also seems to misunderstand the expression '**across**' so it'd rather to be changed.

hamishwillee, 2018-04-11:
Thank you. I've clarified that you blow INTO the pitot in https://github.com/PX4/px4_user_guide/pull/194 - feel free to comment.

bigbellmercy, 2018-04-11:
Thanks a lot, Nice change!

PR #194 (2018-04-13): Clarify how airspeed calibrated
This fixes #193.

@dagar I've set you as reviewer as this uses information you provided in http://discuss.px4.io/t/airspeed-sensor-calibration/5083/3

Specifically I think it is useful to mention all those points you made in above discussion.
1. The "quick test"
2. What calibration is actually trying to achieve (baseline with 0 airspeed/wind).
3. That this can also help you detect swapped ports.

The most important point being that you blow INTO the pitot.
closed 2018-04-13 by hamishwillee