PX4/PX4-user_guide #199: Peripheral Hardware - if supported, document Emild Reach RTK

hamishwillee, 2018-04-16:
As discussed here: https://github.com/PX4/Firmware/issues/7275#issuecomment-381771120

Add page here: [Peripheral Hardware](https://docs.px4.io/en/peripherals/)
Add info/links here: https://docs.px4.io/en/advanced_features/rtk-gps.html

paan2097, 2018-04-17:
This would be greatly appreciated! I'm trying to get this working by Thursday 4/19 for a deadline

hamishwillee, 2018-04-17:
@paan2097 It will not happen by 19/4. Hopefully there might be some advice from
https://px4.slack.com/archives/C3GFT2QS3/p1524004278000211?thread_ts=1524004250.000370&cid=C3GFT2QS3

paan2097, 2018-06-04:
Has anyone gotten the emlid reach rtk working with px4 yet?

moreba1, 2020-10-14:
@hamishwillee 
I think that Emlid reach m+ does not support currntly even for GPS only.

hamishwillee, 2020-10-14:
Thanks @moreba1 
I'm going to close this as "not supported" until someone tells me that it is supported.

closed 2020-10-14 by hamishwillee