PX4/PX4-user_guide #205: Document all failsafes

hamishwillee, 2018-04-19:
Failsafes are a real feature of an autopilot. Worth mentioning in high level concepts. 

Some useful links

https://docs.px4.io/en/advanced_config/parameter_reference.html#COM_LOW_BAT_ACT
https://docs.px4.io/en/advanced_config/parameter_reference.html#COM_POS_FS_DELAY
https://docs.px4.io/en/advanced_config/parameter_reference.html#COM_OBL_ACT
https://docs.px4.io/en/advanced_config/parameter_reference.html#COM_OBL_RC_ACT
https://docs.px4.io/en/advanced_config/parameter_reference.html#MIS_DIST_1WP
https://docs.px4.io/en/advanced_config/parameter_reference.html#MIS_DIST_WPS
https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_DLL_ACT
https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_RCL_ACT
https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_RCL_LT
https://docs.px4.io/en/advanced_config/parameter_reference.html#VT_FW_ALT_ERR
https://docs.px4.io/en/advanced_config/parameter_reference.html#gps-failure-navigation
https://docs.px4.io/en/advanced_config/parameter_reference.html#geofence
https://docs.px4.io/en/advanced_config/parameter_reference.html#CBRK_GPSFAIL
https://docs.px4.io/en/advanced_config/parameter_reference.html#BAT_EMERGEN_THR
https://docs.px4.io/en/advanced_config/parameter_reference.html#GF_ACTION 
https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_TRAFF_AVOID

hamishwillee, 2018-04-20:
Also include this info (from @barzanisar )

> You can also directly configure the most common failsafe actions e.g. what to do in case of low battery, RC loss, Data link loss (between vehicle and QGC) and geofence breach in the QGC's [**Vehicle Setup > Safety**](https://docs.qgroundcontrol.com/en/SetupView/Safety.html). Additionally, the **Safety** tab provides the commonly used Return to Home and Land Mode Settings which are also important for mission mode. Configuring these settings is an easier alternative to changing the corresponding parameters in the **Parameters** tab.

hamishwillee, 2018-04-29:
See https://github.com/mavlink/qgc-user-guide/issues/138 for Battery clarification discussion

hamishwillee, 2018-04-30:
There is a question in https://github.com/mavlink/qgc-user-guide/issues/138

> Critical and emergency battery thresholds parameters state "This threshold commonly will trigger RTL." It would be clearer if it states that it will follow the failsafe selected in the Safety setup. Or am I wrong and there is some override of the desired failsafe, and it goes to RTL in certain instances?

The way this works is that there are 2 battery thresholds (docs indicate "the common action at threshold" as shown):
* [BAT_EMERGEN_THR](https://docs.px4.io/en/advanced_config/parameter_reference.html#BAT_EMERGEN_THR) - Landing
* [BAT_CRIT_THR](https://docs.px4.io/en/advanced_config/parameter_reference.html#BAT_CRIT_THR) - RTL

However the actual action is defined in [COM_LOW_BAT_ACT](https://docs.px4.io/en/advanced_config/parameter_reference.html#COM_LOW_BAT_ACT) and defaults to warning - you get a warning at both thresholds. You can also select RTL or Land (which will obviously be invoked whichever threshold is reached first). Last of all you can select to RTL for critical and Land for emergency. 

The problem is that the docs indicate that you will commonly see the last option, while the actual response defaults to just warning on both thresholds. 

@MaEtUgR IMO EITHER the default action should change to 3 OR the text for the BAT_EMERGEN_THR and BAT_CRIT_THR should change from "This threshold commonly will trigger RTL" to "This threshold will trigger warning, RTL or Land, depending on the value of COM_LOW_BAT_ACT."

hamishwillee, 2018-05-02:
This will be fixed by #212

MaEtUgR, 2018-05-23:
@hamishwillee I think the default is to do no action because if someone builds a prototype vehicle, it doesn't fly properly in every mode yet and/or the battery parameters aren't configured correctly it could automatically do something unsafe (I've experienced this case before). Hence it was probably decided to be better if the vehicle fully obeys on remote input by default.

As you suggest I think the documentation should say it warns and can be configured to react automatically.

MaEtUgR, 2018-05-23:
@hamishwillee Are all the listed conecrns now fixed with https://github.com/PX4/px4_user_guide/pull/212 ?
Then we could close this issue...

hamishwillee, 2018-05-24:
Thanks @MaEtUgR Yes, I'll close this. Thanks for the tip re "fixes". I knew there was some magic, but hadn't realised exactly what it was!

closed 2018-05-24 by hamishwillee