PX4/PX4-user_guide #213: Doc Bug: Flight Modes · PX4 User Guide

bigbellmercy, 2018-05-03:
Hi, in 'Position' and 'Altitude' modes of 'Fixed Wing', there is no description for 'yaw stick'.

So users including me cannot know how the heading is determined and what is the role of yaw stick.

Bug Page: [Flight Modes · PX4 User Guide](https://docs.px4.io/en/flight_modes/)

hamishwillee, 2018-05-04:
Hmmm. Yaw stick controls yaw :-) Do you think we need to add that to the list?

This is very much a summary. At the top of the page you will see this link: https://docs.px4.io/en/getting_started/flight_modes.html

Thoughts?

bigbellmercy, 2018-05-05:
Nice day~ I agree. You’d rather decide to change or only consider it later. I just informed my case.

hamishwillee, 2018-05-11:
Thanks @bigbellmercy Done!

closed 2018-05-11 by hamishwillee

bigbellmercy, 2018-05-11:
Great Hamish!