PX4/PX4-user_guide #244: Multicopter PID Tuning Guide update and generic racer setup page

hamishwillee, 2018-05-28:
[Multicopter PID Tuning Guide · PX4 User Guide](https://docs.px4.io/en/advanced_config/pid_tuning_guide_multicopter.html) needs a major overhaul
- Vehicles not actually tuned in this way. 
- MC_YAW_FF is likely to be removed soon 

Also have generic racer setup page. 

Draft to be done by @bkueng with help from @MaEtUgR and @RomanBapst 

This originates from discussion here: https://github.com/PX4/px4_user_guide/pull/242#issuecomment-391965843

MaEtUgR, 2018-06-11:
@hamishwillee Good idea. You could assign me as well.

bkueng, 2018-06-11:
FYI I started working on a racer page already. I'll probably do the tuning page as well.

MaEtUgR, 2018-06-11:
@bkueng That's perfect, I planned to postpone this if I had to write up myself. Let me know if I can help and/or review anything.

closed 2023-05-24 by hamishwillee