PX4/PX4-user_guide #251: Doc on VIO/Vision systems

hamishwillee, 2018-05-30:
PX4 supports vision fusion/VIO, but this fact is not easily discoverable (there is no obvious/easy way to find out that this is supported and how to go about setting it up). 

Specifically nothing "canonical" shows up in search on vision or VIO and nowhere can you find a statement about our level of support.  There is information about this in [the EKF doc](https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html), but it is buried, and only covers parameters, not hardware. 

So the proposal is a high level doc for this topic, with the following properties:
- Name (?): "Vision-Based/ Indoor Navigation"
- Doc lives in [PX4 User guide > Advanced Features](https://docs.px4.io/en/advanced_features/). Should also be linked from Concepts/Features.
- Doc covers required hardware and links to info on setup
- Doc covers parameters to enable in both LPE and EKF2
- Doc covers technology overview - ie what do you need to do to get what sort of accuracy/behaviour
- Maybe small section on "up and coming features"
- Anything else?

In addition, we'll migrate [the EKF doc](https://dev.px4.io/en/tutorials/tuning_the_ecl_ekf.html) into the PX4 User Guide.

hamishwillee, 2018-05-30:
@bkueng Is there anyone you recommend I talk to about this? In the short term I would like to understand the _terminology_ (ie what is VIO vs "Vision Navigation" and other ways this is referred to), the _technology_ - what we support vs technologies people might expect us to support, where we're good and bad. That should lead me into the detail of exactly what hardware is needed.

At the end of the chain I can then close the loop on the specific estimator features with @priseborough (unless Paul wants want to brief me first - perhaps time to revisit those presentation notes!)

bkueng, 2018-05-31:
> @bkueng Is there anyone you recommend I talk to about this?

Yes, there are a few people: @ChristophTobler @mhkabir @nicovanduijn @TSC21

hamishwillee, 2019-12-19:
This is resolved by having the dev guide page https://dev.px4.io/master/en/advanced/computer_vision.html

Note, this is a dev-guide page. One day we should have a user-guide page, but that kind of depends on computer vision being more turnkey.

closed 2019-12-19 by hamishwillee