PX4/PX4-user_guide #260: PX4 User Guide site appears to be broken

seanrowens, 2018-06-13:
If I google on;

PX4 user guide

google shows me a top hit of;

---
Introduction · PX4 User Guide
https://docs.px4.io/en/
Getting Started should be read by all users! It provides an overview of PX4, including features provided by the flight stack (flight modes and safety features) and ...
---

Clicking on the above link takes me to https://px4.gitbook.io/px4-user-guide which appears to be pretty much empty.  Screenshot attached.

[image]

hamishwillee, 2018-06-13:
Yes it is. We're looking into it. Thanks.

hamishwillee, 2018-06-13:
And for now, you can access current content from here: 
https://hamishwillee.gitbooks.io/havdevguide/content/en/
https://hamishwillee.gitbooks.io/ham_px4userguide/content/

seanrowens, 2018-06-13:
Many thanks, that helps.

hamishwillee, 2018-06-14:
Fixed! Thanks for reporting this.

closed 2018-06-14 by hamishwillee

seanrowens, 2018-06-14:
Thanks for making some awesome open source software and (especially!) docs!