PX4/PX4-user_guide #266: Jenkins add htmlproofer

dagar, 2018-06-15:
@hamishwillee can you review what you want here and we'll get it added to Jenkins.

	-- htmlproofer ./_book 	
	-  --empty-alt-ignore true 	
	-  --check-external-hash true	
	-  --url-ignore 'https://img.shields.io/badge/discuss-px4-ff69b4.svg,https://img.shields.io/github/release/PX4/Firmware.svg,https://support.intrinsyc.com/attachments/download/1571/Flight_3.1.3.1_qcom_flight_controller_hexagon_sdk_add_on.zip,https://store.3dr.com/products/3dr-pixhawk,https://store.3dr.com/products/433-mhz-telemetry-radio,https://store.3dr.com/products/915-mhz-telemetry-radio,https://store.3dr.com/products/digital-airspeed-sensor,https://store.3dr.com/products/wifi-telemetry-radio'

hamishwillee, 2018-06-18:
@dagar What I need here is an external link checker that I can use to get a report on all broken links in a build, but NOT fail the build. I will then on occasion go through the links and fix them up. 

htmlproofer is a link checker that integrates nicely with Travis and Gitbook output. You can use whatever is easiest for you with Jenkins. Does that answer the question?

dagar, 2018-06-18:
Why shouldn't it fail the build? Isn't it currently clean?

hamishwillee, 2018-06-19:
@dagar Because external links can break at any point - unrelated to a particular PR or anything we are doing. I don't want to block work being submitted due to unrelated issues, but I do want an easy way to check external links.

lamping7, 2018-06-22:
If it doesn't break the build, how will we know what needs fixed?

hamishwillee, 2018-06-22:
Well on the old version, I just go review the travis build log. That shows any failed links, but does not stop the build from completing and being marked as failed for the purposes of merging.

hamishwillee, 2023-05-24:
Closing this as "almost done".

I have a written and deployed a custom markdown link checker. This currently only checks internal links. It is aware of our document structure, and use of anchors etc, and can report lots of other things like orphan files.
Most important, it is aware of what files are changed, so it can report just those issues that are relevant to the poster of a PR.

I still have some work to do - support for external link checking, and ability to hide errors that we know are artifacts of the setup or external site.

closed 2023-05-24 by hamishwillee