PX4/PX4-user_guide #294: Lidar-Lite misleading note for the Intel Aero

IeiuniumLux, 2018-07-17:
Hi Hamish,

The [Lidar-Lite page](https://github.com/PX4/px4_user_guide/blob/67745a094608bcaaca9c4452f3b078270d4083ad/en/sensor/lidar_lite.md#lidar-lite) has a little bit misleading note for the Intel Aero. The note links you to the Intel Aero page just to find out a [waring](https://github.com/PX4/px4_user_guide/blob/df32443e5fbb14240270a26e93526f7ddeccfd0b/en/flight_controller/intel_aero.md#connecting-lidar-lite-range-finder-lidar_lite) actually not recommending to use it with the Intel Aero. If the warning for the Intel Aero is correct, then it's probably better to add it to the Lidar-Lite page as well.

closed 2018-07-25 by hamishwillee

IeiuniumLux, 2018-07-25:
@hamishwillee sorry, I was out for a few days.  Yes, that is what I meant regarding the connector being flipped over so you got it.  The pins are on opposite sides in v3 unit. Thanks!

hamishwillee, 2018-07-25:
Thanks Al, I've made that clear now on master.

PR #295 (2018-07-25): Update lidar_lite connection/wiring info
- Fixes #294
- Fixes up docs that suggest you connect to Aux "row 6", when means "column"
- Adds info about LidarLite pinout - so can be used for I2C connections by people who know how to wire I2c. Also adds the garmin spec for v3 and the standard wiring images from spec.
- Clarifies that v3/2 pinout is the same.

Open question 
- Pinout v2/3 is same, but physical layout of what is pin 1 and pin 6 may have flipped. Confirm. 
- v3 advice is 1KOhm pulldown resister while v2 advice is 470Ohms. Does it matter? What should we recommend?