PX4/PX4-user_guide #319: Doc Bug: Airspeed · PX4 User Guide

IeiuniumLux, 2018-08-14:
[Step 4](https://github.com/PX4/px4_user_guide/blob/master/en/config/airspeed.md#performing-the-calibration) is very misleading/confusing with that warning, don't you think so?

Bug Page: [Airspeed · PX4 User Guide](https://docs.px4.io/en/config/airspeed.html)

hamishwillee, 2018-08-15:
@IeiuniumLux The point is to "shield" the sensor from direct wind but not it is important not to block the sensor holes. The warning is therefore necessary - because otherwise people might block the holes.

Would you find it clearer as:

> Shield the sensor from the wind (i.e. cup it with your hand, while taking care not to block any holes).

IeiuniumLux, 2018-08-15:
Hi Hamish, As usual, thank you for the quick follow up.  The way you explained above makes more sense.  I just made small changes.

> Shield the sensor from the wind (e.g. cup it with your hand); while taking care of not blocking any of the holes.

hamishwillee, 2018-08-16:
Thanks! Fixed. Further slight modification.

commit e4f8f81d (2018-08-16): Clarify how pitot is protected from wind.

Fixes #319
closed 2018-08-16 by hamishwillee