PX4/PX4-user_guide #326: Add better camera doc

hamishwillee, 2018-08-27:
This is copied from https://github.com/PX4/px4_user_guide/pull/49 (merged PR that needs more work)

We need documents in user guide on "how to set up cameras". Once complete, this will be it. I've "seeded" it by copying [Camera Trigger](https://dev.px4.io/en/tutorials/camera_trigger.html) tutorial from the devguide.

This will have to cover:
1. Overview of what is covered - integration of still camera for use during missions/survey (and also available for manual triggering). Not video/FPV (link to that if we have docs)
2. Explanation of mechanisms for physically connecting cameras
   - No info in any docs, but "hints" in parameters
     1: GPIO
     2: Seagull MAP2 (over PWM)
     3: MAVLink (forward via MAV_CMD_IMAGE_START_CAPTURE)
     4: Generic PWM (IR trigger, servo)
3. Lists of known good/tested /recommended cameras
4. Explanation of how to configure cameras (ie the parameters)
5. Link to other information on *using* cameras.
6. Links to MAVLink camera API.

Note:
- How to use a camera in a mission would be a separate "mission" based document. We would link to it from here
- The camera mission interface would probably be here. 
- POSSIBLY this should also include gimbal integration.

hamishwillee, 2018-08-27:
Make sure any info required for Camera Capture driver is added: https://github.com/PX4/Firmware/pull/10295#issuecomment-416093746

IeiuniumLux, 2019-01-31:
@hamishwillee, Currently there are two Camera docs online with the same title "Camera Trigger".  The one you mention [above](https://docs.px4.io/en/peripherals/camera.html) and another one, but with less content [here](http://dev.px4.io.s3-website-us-east-1.amazonaws.com/advanced-camera-trigger.html#).  It might be a good idea to retire the latter.

hamishwillee, 2025-01-29:
This section all got tidied last year. Closing.

closed 2025-01-29 by hamishwillee