PX4/PX4-user_guide #332: Probably should cover telemetry ports/writing in radios

hamishwillee, 2018-09-12:
This [mavlink uart](https://github.com/mavlink/c_uart_interface_example#exploration) example states:

>First you can connect via a Telemetry Radio on Telem 1 or 2, or via an FTDI on Telem 2 or Serial 4 (https://pixhawk.org/dev/wiring). Note: Serial 5's receive pin is occupied by a second NuttX shell and can't be used to receive data without reconfiguration.

The link to pixhawk.org no longer exists, but clearly we do need to have a link which says how you connect a telemetry radio. 

I was thinking perhaps we should merge [MAVLink (OSD / Telemetry)](https://docs.px4.io/en/peripherals/mavlink_peripherals.html) into [Telemetry Radios · PX4 User Guide](https://docs.px4.io/en/telemetry/) and discuss both settings and wiring as part of the new port setting updates.

@bkueng Would appreciate your thoughts on this.

bkueng, 2018-09-12:
> I was thinking perhaps we should merge MAVLink (OSD / Telemetry) into Telemetry Radios · PX4 User Guide and discuss both settings and wiring as part of the new port setting updates.

Merging makes sense.
What you need in addition to the provided info on the uart example page is the pinout (and the Telemetry if you want to test with that. But the first step is probably via FTDI).

hamishwillee, 2018-09-13:
>> I was thinking perhaps we should merge MAVLink (OSD / Telemetry) into Telemetry Radios · PX4 User Guide and discuss both settings and wiring as part of the new port setting updates.

> Merging makes sense.

I've reconsidered this. The work you've done to split the TELEM ownership by per module params means that we should talk about the functions "OSD" and "Telem radio" rather than port ownership.  UPshot, I think that having an OSD page makes sense, but we will have to make one that is meaningful - the current version is pointells.

hamishwillee, 2018-09-13:
> What you need in addition to the provided info on the uart example page is the pinout (and the Telemetry if you want to test with that. But the first step is probably via FTDI).

I'm not sure I know enough to do this.

Telemetry radios are easy because there is no wiring up. I guess we just point to the telemetry page, which should explain how to set params for the port?

For FTDI I think you're suggesting we need to point how to find map the RTS, RX, TX, 5V, CTS, GND on the cable (https://docs.px4.io/en/flight_controller/pixhawk.html#pinouts) to the serial 4 pinouts of the serial 4/5 port - is that correct?

- Do we just connect the TX, RX GND and 5V - ie ignore the RTS and CTS on the cable?
- Does the pinout mapping change across Pixhawk boards?
- How do we have to setup the params? I guess for serial 4/5 we don't have to do anything, but if we were to connect to TELEM2 we would need to set SYS_COMPANION to 0 (disabled?)

bkueng, 2018-09-13:
> Telemetry radios are easy because there is no wiring up. I guess we just point to the telemetry page, which should explain how to set params for the port?

Yes in that case it's enough. When you work with UARTs you might have an FTDI available, but not necessarily a Telemetry.

> For FTDI I think you're suggesting we need to point how to find map the RTS, RX, TX, 5V, CTS, GND on the cable (https://docs.px4.io/en/flight_controller/pixhawk.html#pinouts) to the serial 4 pinouts of the serial 4/5 port - is that correct?

Correct.

> Do we just connect the TX, RX GND and 5V - ie ignore the RTS and CTS on the cable?

Yes RTS and CTS are not required.

> Does the pinout mapping change across Pixhawk boards?

I think for most of them it's the same. But I cannot say for sure.

> How do we have to setup the params? I guess for serial 4/5 we don't have to do anything, but if we were to connect to TELEM2 we would need to set SYS_COMPANION to 0 (disabled?)

Exactly, you need to make sure to disable what's already running on the port you want to use. This will become possible with the new configuration params or via SYS_COMPANION on TELEM2.

hamishwillee, 2018-09-14:
Thanks very much. Sorted by https://github.com/mavlink/c_uart_interface_example/pull/30

closed 2018-09-14 by bkueng