PX4/PX4-user_guide #333: Some modifications might be needed in Crazyflie 2.0 page.

Xiaoyu0601-Wang, 2018-09-12:
In the "Crazyflie 2.0 -> Wireless Setup Instructions" part, 
if I follow the instructions of installation, the cfbridge will work without showing  
"waiting to receive message
received 17 bytes from ('127.0.0.1', 14550)", and QGC cannot detect crayflie.

I think the problem here is that the python access permission to usb port is denied.

So I deleted all pip installations and virtualenv, and added the "--user" to all pip installations.
Then I can receive data from usb port, and I can connect with QGC.

I'm not sure whether this modification is necessary or not. But the connection with QGC via crazyradio is good now in my case.

bkueng, 2018-09-12:
@barzanisar ?

barzanisar, 2018-09-12:
I didn’t across this issue when I was setting up cfbridge but since “— user” option installs packages into the default Python environment, it’ll be nice to add it in the crazyflie px4 guide. Can you please open a PR for it?

hamishwillee, 2018-09-12:
I created and merged PR for this https://github.com/PX4/px4_user_guide/pull/334. `--user` causes no harm.

Thanks @Xiaoyu0601-Wang .

closed 2018-09-12 by hamishwillee