PX4/PX4-user_guide #368: Compass - clarify how multiple compasses work

hamishwillee, 2018-10-17:
1. Clarify conditions under which internal compass might be used
   - Recommendation is *always* use an external magnetometer if you can (usually combined with the GPS) mounted as far from other electronics as possible. The reason for this is that compasses are susceptible to electromagnetic noise, so the further from other electronics the better. An internal compass will always be close to the other pixhawk electronics, so potentially much less reliable.
   - Use internal when weight really matters and having accurate heading/compass information does not. Essentially that means racers, where most of the time you're running in some sort of acro mode with a lot less autopilot support.
1. Clarify how multiple compasses work on PX4 (assumptions on how this works below). 
   - How/whether you calibrate "all" compasses
   - How compass inputs are used - fused, or just one selected. If one is selected, what are rules.

Affected docs:
- https://docs.px4.io/en/gps_compass/ 
- https://docs.px4.io/en/config/compass.html
- https://docs.px4.io/en/getting_started/sensor_selection.html#gps_compass

hamishwillee, 2018-10-17:
How I think compasses work.

Looking at the params:
- we support up to 4 magnetometers, which are enabled using CAL_MAGX_EN, where X=0-3. You can disable any of them using this parameter, but you must have at least one.
- YOu can specify them as internal or external. This should be done for you automatically by the driver for internal ones though.
- There is a primary compass which is identified using CAL_MAG_PRIME. This is the only compass used by EKF fusion.

Some questions:
1. Is above correct?
2. QGC only appears to allow you to calibrate one compass (ie it doesn't allow you to point at specific compasses). Is it actually calibrating all of them under the hood, or just the prime one? If just prime, how do the others get calibrated?
3. How do things get to be "Prime" is it just a matter that first external is prime with ordered fallback to internal or is there some other rule?

hamishwillee, 2018-10-17:
@dagar @bkueng Anything you can add in terms of above info?

bkueng, 2018-10-18:
> Recommendation is always use an external magnetometer if you can (usually combined with the GPS) mounted as far from other electronics as possible.

Correct.

> The reason for this is that compasses are susceptible to electromagnetic noise, so the further from other electronics the better. An internal compass will always be close to the other pixhawk electronics, so potentially much less reliable.

It's mostly about strong currents, so current from the battery to the motors. This means an internal compass might be usable on a larger vehicle (e.g. VTOL), where you can place the Pixhawk far enough away from the motor/ESC power supply lines.

> Essentially that means racers, where most of the time you're running in some sort of acro mode with a lot less autopilot support.

No! :) On small vehicles the internal compass is pretty much always useless. On a racer w/o external compass you better not use the mag at all.
The heading is usually not just a bit less accurate, but I've seen cases where when you throttle up, the vehicle yaws about 90 degrees due to interference.

> we support up to 4 magnetometers, which are enabled using CAL_MAGX_EN, where X=0-3. You can disable any of them using this parameter, but you must have at least one.

Correct.

> YOu can specify them as internal or external. This should be done for you automatically by the driver for internal ones though.

You should not change that however. The parameter helps to see which one is external.

> How/whether you calibrate "all" compasses

All compasses are automatically calibrated.

> There is a primary compass which is identified using CAL_MAG_PRIME. This is the only compass used by EKF fusion.

Correct. Internally they have a priority and the highest priority is selected. External mags have high priority, so they are prefered automatically. If it fails in-flight, it will failover to the next one.
If it failed before flight, arming will be denied.

> Is it actually calibrating all of them under the hood, or just the prime one? If just prime, how do the others get calibrated?

It will only show the rotation for all external mags. Which is why you only see the one.