PX4/PX4-user_guide #373: Add "cut-in" date and next release for a feature.

davids5, 2018-10-31:
@hamishwillee - On the dev-call we discussed this and @dagar asked me to post this:

 We have an opportunity to reduce some confusion that stems from a user using old code with current docs by adding a 'cut-in' date of when a documented change occurred and in what stable release it will be incorporated.

hamishwillee, 2018-11-01:
Hi @davids5 @dagar

Yes, it is a legitimate concern. The other concern is that they might still want those old docs.

There is really only one _effective_ way to address this problem - add versioning to the documentation set. So master remains as is and we take a copy of the docs for each release. I use this approach for the [SDK](https://sdk.dronecode.org/en/):
[image]

Gitbook.com does this using either a branch or a tag. You use a branch if you want to be able to add docs to older versions of the library, or if you don't know what release a feature is going into. You use a tag if all docs features are all going into the next release (this is my preference).

The alternative is to add some sort of note whenever you add a feature. However that requires a lot more planning, doesn't give you easy access to the legacy docs, and will miss a whole lot of changes like parameter updates.

@dagar It is done with this plugin: https://plugins.gitbook.com/plugin/versions
Obviously you would have to create the legacy copies when we make a doc release and then point to them using the "manual" setup in that link.

Does that make sense?

hamishwillee, 2025-01-29:
Closing this as stale/addressed.

The current docs are versioned. We have also started adding badges in main, such as this one at the top of the page, showing when things were introduced: https://docs.px4.io/main/en/ros2/px4_ros2_interface_lib.html#px4-ros-2-interface-library

I'd like it if we could use that more widely to indicate introduction of features, including new drivers etc.

closed 2025-01-29 by hamishwillee