PX4/PX4-user_guide #378: Doc about MindPX firmware

airmind, 2018-11-12:
@hamishwillee i noticed docs on MindPX hardwares are re-organized and looks good, thanks for the work.

One problem i found on https://docs.px4.io/en/flight_controller/mind_series.html is, as quoted below, 

> At time of writing, MindPX firmware is not available for download through QGroundControl. To use these boards you will have to build the firwmare yourself (using the same NuttX toolchain as Pixhawk.

I am not sure i correctly understand this text. Is it stating that PX4 on MindPX hardware can not downloadable through QGC now? as i did not see such changes on QGC side. 

Or it means customized mindpx firmware?

thanks.

hamishwillee, 2018-11-12:
@airmind What it means is that MindPX uses a custom build of firmware that isn't automatically available for download from QGC backend servers. If you want to use MindPX you have to build and upload your own firmware. You can do this through QGC as "custom firmware"

Does that make sense?

IF so, I can improve this text to : "At time of writing, MindPX firmware is not automatically available for download through QGroundControl (but can be installed as "Custom Firmware"). To use ..."

Would that be better?

airmind, 2018-11-14:
@hamishwillee thanks for the reply.

I think there are some misunderstanding, as MindPX firmware is one of the targets of official PX4 releases, and can actually be downloaded/upgraded through QGC automatically.

This statement probably a historical text? been written some time back long ago before MindPX board was supported by QGC. It should be obsolete now, and probably just remove it in my suggestion.

Thanks a lot.

hamishwillee, 2018-11-15:
Thanks @airmind - yes, historical text. I have removed note in the page (see linked changelist).

closed 2018-11-15 by hamishwillee

airmind, 2018-11-16:
Thank you @hamishwillee !