PX4/PX4-user_guide #382: Reduce Control Latency

IeiuniumLux, 2018-11-21:
In the [Control Latency](https://github.com/PX4/px4_user_guide/blob/7d13e9c39a17f12a43514fd52e9cdfa783a26e31/en/config_mc/racer_setup.md#control-latency-control_latency) section, this line should read:

"These are the factors that contribute to **_reduce_** the latency:"

Instead of:

"These are the factors that contribute to the latency:"

Without the word **_reduce_**, the reader might interpret that the bullets belows contribute to the latency; when in reality they reduce the latency.

hamishwillee, 2018-11-22:
Thanks! Proposed fix in #384

bkueng, 2018-11-22:
That is not correct, they all in fact contribute to increasing the latency. For example a softer mounted airframe will add latency.

hamishwillee, 2018-11-22:
Thanks @bkueng 

There is still a problem here, because @IeiuniumLux was able to misinterpret the text. I think it is because the points refer to reducing latency even though the introduction says contribute to.

How about we change FROM
```
These are the factors that contribute to the latency:"
- A soft airframe or soft vibration mounting (they act as a filter).
- Low-pass filters in software and on the sensor chip.
- PX4 software internals: the sensor signals need to be read in the driver and then pass through the controller to the output driver.
  Enable [SYS_FMU_TASK](../advanced_config/parameter_reference.md#SYS_FMU_TASK) to reduce latency (default on [Pixracer](../flight_controller/pixracer.md) and [Omnibus F4 SD](../flight_controller/omnibus_f4_sd.md)).
- The IO chip (MAIN pins) adds about 5.4 ms latency compared to using the AUX pins (this does not apply to a *Pixracer* or *Omnibus F4*, but does apply to a Pixhawk).
  To avoid the IO delay, disable [SYS_USE_IO](../advanced_config/parameter_reference.md#SYS_USE_IO) and attach the motors to the AUX pins instead.
- PWM output signal: enable One-Shot ([PWM_RATE](../advanced_config/parameter_reference.md#PWM_RATE)=0) to reduce latency.
```
TO (use "affect" rather than contribute, and then clarify in specific cases what they do)
```
These are the factors that affect the latency:
- A soft airframe or soft vibration mounting increases latency (they act as a filter).
- Low-pass filters in software and on the sensor chip trade off increased latency for improved noise filtering.
- PX4 software internals: the sensor signals need to be read in the driver and then pass through the controller to the output driver.
  Enable [SYS_FMU_TASK](../advanced_config/parameter_reference.md#SYS_FMU_TASK) to reduce latency (default on [Pixracer](../flight_controller/pixracer.md) and [Omnibus F4 SD](../flight_controller/omnibus_f4_sd.md)).
- The IO chip (MAIN pins) adds about 5.4 ms latency compared to using the AUX pins (this does not apply to a *Pixracer* or *Omnibus F4*, but does apply to a Pixhawk).
  To avoid the IO delay, disable [SYS_USE_IO](../advanced_config/parameter_reference.md#SYS_USE_IO) and attach the motors to the AUX pins instead.
- PWM output signal: enable One-Shot to reduce latency ([PWM_RATE](../advanced_config/parameter_reference.md#PWM_RATE)=0) 
```

bkueng, 2018-11-23:
Yes the change is good.

hamishwillee, 2018-11-26:
Thanks guys, committed change as https://github.com/PX4/px4_user_guide/commit/5332ad444c5a000b11d754eef441027381c7a212

closed 2018-11-26 by hamishwillee