PX4/PX4-user_guide #397: Add Orbit Mode

hamishwillee, 2018-12-05:
Need to document the Orbit mode. 

Some key links
- [Orbit: Send telemetry for the UI #10989](https://github.com/PX4/Firmware/pull/10989)
- [Enable Orbit Flight Mode](https://github.com/PX4/Firmware/pull/10907)
- [MAV_CMD_DO_ORBIT](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_ORBIT)
- [ORBIT_EXECUTION_STATUS](https://mavlink.io/en/messages/common.html#ORBIT_EXECUTION_STATUS)

As I understand it, 
- this "mode" is initiated by the [MAV_CMD_DO_ORBIT](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_ORBIT). 
  - specify circle 3D centre point, radius, tangential velocity (speed and direction), and yaw (direction vehicle faces)
  - It says "Center point latitude (if no MAV_FRAME specified) / X coordinate according to MAV_FRAME."
    - In the second case it says X co-ordinate in MAV_FRAME (which is clear). But the first cases is "Center point latitude (if no MAV_FRAME specified)" 
      - but that doesn't tell us the frame if no MAV_FRAME is specified? 
      - Is that only a concern if mission is sent in COMMAND_LONG?
- While flying the mode vehicle should output [ORBIT_EXECUTION_STATUS](https://mavlink.io/en/messages/common.html#ORBIT_EXECUTION_STATUS)
  - publishes 3D centre point, radius and travel direction.

Questions:
- Release - when will this all go live? Ie when in master, and when expected in a release?
- Can this only be entered via the command? 
  - ie there is no way to invoke this from an RC switch?
- This is a true mode like Acro etc right? ie not just a mission command.
- Can it be run in a mission?
- What path does the vehicle fly to start the circle? Just straight line to closest edge at given altitude, or is there some pretty path? Does it matter?
- Are there any parameters that affect the behaviour? (ie can it be tuned)
- Does it work for VTOL in MC mode.
- What is the expected UI interaction? 
  - My understanding is that user will only be able to set centre point with default radius on QGC, and then will be able to drag the radius in and out.
   - Will they also be able to change speed and direction from UI?
   - This is another case of where I can imagine a user might want to have control from RC. On Solo for example you can set the ROI point and then orbit around the point, changing the radius with your joystick.

hamishwillee, 2018-12-06:
@MaEtUgR Thanks. Firstly, the main open question was this bit:

Currently it says 
> Center point latitude (if no MAV_FRAME specified) / X coordinate according to MAV_FRAME.

But I think you are saying it should say:
> Center point latitude. X coordinate according to MAV_FRAME (or MAV_FRAME_GLOBAL if no MAV_FRAME specified).

Is that correct? Basically we either need to define a default frame for this in MAVLink spec OR do whatever is normally done for COMMAND_LONG (I have no idea!)

note, we can now specify default= attributes, which are used only by GCS.

hamishwillee, 2018-12-06:
>> Release - when will this all go live? Ie when in master, and when expected in a release?

> Yuneec will have it in the next bigger H520 release. For master you just need PX4/Firmware#10907 to be merged for starting it and PX4/Firmware#10989 to have the telemetry sent out. QGC UI for the telemetry is WIP by @DonLakeFlyer . 
> So I assume it will also be available in the next PX4 release right?

One would hope. I'm not sure how the releases are managed. I did see Ramon asked you if this would likely be ready for 1.9.

>> Can this only be entered via the command?

> Currently yes, enabling it by RC won't do anything. We could in theory enable it also by RC and have some default parameters e.g. center that is x amount of meters north from the vehicle, radius x and 1m/s speed. But I tried to use that in reality and I think in practise it's totally useless because you cannot change the center without the command and it's very tedious to configure the circle this way...

Solo UI is demoed here: https://youtu.be/JYFXOg0vt2U?t=388

Basically centre is set in GCS, but RC then allows you to change the radius and the direction.

>> Does it work for VTOL in MC mode.

> Theoretically yes. I didn't and can't try. Is there a use case?

For docs it doesn't matter - I just like to say what it does :-)

But I would say is the same as for MC - you've flown to far destination using FW mode, now you want to do close inspection orbit using MC mode. Of course you could do a loiter/hold in FW too, but it wouldn't give you the control over camera/yaw you get from this. 

Thanks very much!

DonLakeFlyer, 2018-12-06:
@hamishwillee is correct with his comments. In reality MAV_FRAME makes a lot of sense in the gcs->vehicle direction as used for mission commands. Since the user is in control of the information being sent. Whereas for vehicle->gcs telemetry I don't think MAV_FRAME makes as much sense. I think you should just spec it to be a single set of specific values. The only downside of that is that it means the telemetry only works in a GPS frame of reference (like say no GPS and optical flow only). But fancy features without a GPS doesn't seem that important.

I don't know if there are examples of other vehicle->gcs telemetry messages which have already ran into this problem or not.

hamishwillee, 2018-12-06:
It is certainly possible and even desirable to restrict the supported frames for a message - they can always be extended later if needed.

hamishwillee, 2018-12-06:
See also https://github.com/mavlink/qgroundcontrol/pull/7066

MaEtUgR, 2018-12-07:
> Basically we either need to define a default frame for this in MAVLink spec

Yes a clear default would makes sense. @DonLakeFlyer any preference? ok with `MAV_FRAME_GLOBAL`?

> One would hope.

Sure, I'm pushing for this feature now that I'm able to. I think it makes sense to be included in 1.9.

> Solo UI is demoed

Interesting, I've never seen that before.

> Basically centre is set in GCS, but RC then allows you to change the radius and the direction.

This statement also holds for our implementation. Just RC input is handeled a bit different. I felt like holding the stick all the time just to continue the orbit is not an option... let's see, that should be easy to adjust with more feedback.

> you've flown to far destination using FW mode, now you want to do close inspection orbit using MC mode

valid idea, yeah that should work just fine as long as you don't go too fast and have some unwanted side effects with your wings.

> Thanks very much!

Thank you for your thoughtfulness.

> spec it to be a single set of specific values. The only downside of that is that it means the telemetry only works in a GPS frame of reference (like say no GPS and optical flow only).

I also thought about that and you name the reason why the message contains the frame. The idea is that the telemetry frame depends on the frame in which the command was.

> But fancy features without a GPS doesn't seem that important.

Well not today but I don't want to limit if it's that easy to have an implementation that is more clear and future proof.

> It is certainly possible and even desirable to restrict the supported frames for a message - they can always be extended later if needed.

I think that's the essential point. The Firmware currently also just supports the case that I'm testing here with @DonLakeFlyer .

hamishwillee, 2018-12-10:
Great to all that. Can you go back and add the changes proposed to the MAVLink message?

PR #472 (2019-05-09): Orbit mode docs
Fixes #397 

This is first draft of Orbit mode docs. Some things are known "not great". Still need to verify that the mode will be present in v1.9.0

Rendered version [here](https://hamishwillee.gitbooks.io/ham_px4userguide/content/v/orbit_mode/en/flight_modes/orbit.html). Some other references [here](https://hamishwillee.gitbooks.io/ham_px4userguide/content/v/orbit_mode/en/getting_started/flight_modes.html#orbit_mc) and [here](https://hamishwillee.gitbooks.io/ham_px4userguide/content/v/orbit_mode/en/flight_modes/#orbit_mc).

@MaEtUgR can you please review?
closed 2019-05-09 by hamishwillee