PX4/PX4-user_guide #405: No .bat in /PX4 directory installing Cygwin

tuloski, 2018-12-13:
The deocumentation here (https://dev.px4.io/en/setup/dev_env_windows_cygwin.html) says that I should have tree .bat in the installation directory:
 - run-console.bat
 - run-eclipse.bat
 - run-vscode.bat

But I actually have only "run-console.bat". 
I don't know if it was a problem during the installation of Cygwin or the documentation is missing something, for example some installation that will put the other two .bat.

hamishwillee, 2018-12-13:
@tuloski YOu didn't do anything wrong. @MaEtUgR removed the links (for now). I'll tidy the docs.

hamishwillee, 2018-12-13:
Thanks again. This will be fixed by https://github.com/PX4/Devguide/pull/695

MaEtUgR, 2018-12-19:
The scripts are not part of the installation anymore. It's because before I had the whole eclipse IDE inside the installer which made it way too bulky and the vs-code script was just a test without any warranty. What I'm currently doing to use the toolchain is installing VS code independently on my Windows machine and setting `"terminal.integrated.shell.windows": "C:\\PX4\\run-console.bat"` in my firmware repository workspace settings. I could actually commit that setting to the PX4 repository such that it's automatically available for every Windows VS Code user. It allows to use the PX4 console conveniently from inside the IDE. Other cmake, compiler, debugging binding to the IDE I didn't get around to test yet.

@tuloski Does that answer the question?

tuloski, 2018-12-19:
As VS Code user I'd be happy to have that, but running run-console.bat and send a command line "make ..." is not really a big issue.
For me it's fine as it is...just change the docs to reflect the actual process ;).

hamishwillee, 2018-12-19:
@MaEtUgR Docs will be fixed once you review https://github.com/PX4/Devguide/pull/695

hamishwillee, 2018-12-24:
Closed by https://github.com/PX4/Devguide/pull/695

closed 2018-12-24 by hamishwillee