PX4/PX4-user_guide #412: Add BetaFPV Beta75X 2S Brushless Whoop Docs

hamishwillee, 2019-01-07:
Add docs for [BetaFPV Beta75X 2S Brushless Whoop](https://www.getfpv.com/beta75x-2s-brushless-whoop-micro-quadcopter-xt30-frsky.html)
- Along the lines of https://docs.px4.io/en/flight_controller/bebop.html ? i.e. 
  - how to build for it, 
  - How to connect to it
  - how to upload new software/new bootloaders
  - How do debug it
  - Any special config
  - How to report bugs for it

Discussion originates in https://github.com/PX4/px4_user_guide/pull/411

hamishwillee, 2019-01-07:
What do you think of the idea of putting this along with aero, crazyflie, bebop under a "Commercial Platforms" or similar super folder. Ie they all share the fact that they are not boards but full vehicles, albeit some are reference platforms.

bkueng, 2019-01-08:
Yes I think something like "Complete Vehicles" makes sense - full/ready-to-fly vehicles where you can flash upstream PX4.
On that note we should also add the [Teal One](https://tealdrones.com/teal-one/) (@dakejahl).

hamishwillee, 2025-01-29:
This work was done at the time (e.g. https://docs.px4.io/main/en/complete_vehicles_mc/betafpv_beta75x.html#betafpv-beta75x-2s-brushless-whoop ) . TealOne does not appear to have been documented.

Closing.

closed 2025-01-29 by hamishwillee