PX4/PX4-user_guide #417: add/review serial port mapping for all boards

dagar, 2019-01-16:
https://github.com/PX4/Firmware/pull/11044

Until recently PX4/Firmware carried simple serial port mapping documentation in the init scripts. https://github.com/PX4/Firmware/blob/82aa24adfca29321cfd1209e287eab6c2b16780e/ROMFS/px4fmu_common/init.d/rcS#L16-L40

After the board reorganization it was decided this should exist only in the official user guide documentation.

Let's go through each currently supported board and make sure equivalent serial port mapping exists in the user guide.

https://docs.px4.io/en/flight_controller/
 
 - [ ] pixhawk 1
 - [ ] pixracer 
 - [ ] pixhawk 2
 - [ ] pixhawk 3 pro
 - [ ] pixhawk 4
 - [ ] pixhawk 4 mini
 etc

davids5, 2019-01-16:
```
# UART mapping on OMNIBUSF4SD:
#
# USART1		/dev/ttyS0		SerialRX
# USART4		/dev/ttyS1		TELEM1
# USART6		/dev/ttyS2		GPS
#
# UART mapping on FMUv2/3/4:
#
# UART1			/dev/ttyS0		IO debug (except v4, there ttyS0 is the wifi)
# USART2		/dev/ttyS1		TELEM1 (flow control)
# USART3		/dev/ttyS2		TELEM2 (flow control)
# UART4
# UART7							CONSOLE
# UART8							SERIAL4
#
#
# UART mapping on FMUv5:
#
# UART1			/dev/ttyS0		GPS
# USART2		/dev/ttyS1		TELEM1 (flow control)
# USART3		/dev/ttyS2		TELEM2 (flow control)
# UART4			/dev/ttyS3		TELEM4
# USART6		/dev/ttyS4		TX is RC input from sbus_RC connector *
# UART7			/dev/ttyS5		PX4IO if present
# UART8			/dev/ttyS6		CONSOLE

```

*From a HW perspective USART6 could be used as a third telem port with HW handshaking *OR* for Serial based RC protocols (SBUS,SUMD...).  In the case as built for pixhawk 4 Hardware the RX, RTS and CTS are not connected. The TX is used as an RX in Single wire mode. This is to enable support for SBUS2 which is not supported as of this writing in FW.

When the PX4IO is not present the FMU (F7) manages the RC and Safety features. When the PX4IO is present the FMU initializes this pins to benign states. Fail over was envisioned but not implemented.

hamishwillee, 2019-01-31:
Don't forget this one gentlemen

davids5, 2020-02-28:
@dagar did this make it somewhere?

PR #672 (2020-03-25): Add serial port mapping
Fixes #417

- 3DR Pixhawk 1 (FMUv2)
  - [x] Done
  - [ ] Reviewed 
 - mRo Pixhawk (FMUv2)
   - [x] Done
   - [ ] Reviewed 
 - mRobotics-X2.1
   - [x] Done
   - [x] Reviewed 
 - HobbyKing HKPilot32 (FMUv2)
   - [x] Done
   - [ ] Reviewed 
 - Holybro Pixfalcon (FMUv2)
   - [x] Done
   - [ ] Reviewed 
 - Drotek Dropix (FMUv2)
   - [x] Done
   - [ ] Reviewed 
 - mRo Pixracer (FMUv4)
   - [x] Done
   - [x] Reviewed 
 - Hex Cube (Pixhawk 2) (FMUv3)
   - [x] Done
   - [ ] Reviewed 
 - Drotek Pixhawk 3 Pro (FMUv4pro)
   - [x] Done
   - [ ] Reviewed 
 - CUAV Pixhack v3 (FMUv3)
  [... cut]
closed 2020-03-25 by hamishwillee