PX4/PX4-user_guide #434: Vibration Management

hamishwillee, 2019-02-06:
Discussion comes from here: https://github.com/PX4/px4_user_guide/pull/427#issuecomment-459586822

With respect to vibration damping it would be great to have more guidance. Ideally a detailed page or two, but if not, links to external resources:

* Are there any good resources external we should link to?
* Do we have anyone here expert enough to write these for our systems?
* Minimally, would it be useful to link the vibration isolation text to the wikipedia article: https://en.wikipedia.org/wiki/Vibration_isolation ?

hamishwillee, 2019-02-06:
> @bresch @priseborough @RomanBapst

Specifically Beat wrote some high level guidance on spotting vibration issues using logs, and included this section on what you can do if you have vibration: https://docs.px4.io/en/log/flight_review.html#solutions
We also suggest "following manufacturer instructions for mounting" in our mounting the flight controller page.

It is a little thin, so I would like to either have more guidance on what you can do, or link to more detail guidance. Can you advise and** can you help?**

If the case is that this is not useful and/or not important, then let me know that too.

hamishwillee, 2019-02-06:
> Response from @RomanBapst 

@hamishwillee @bkueng maybe we can change the formulation from Vibration "damping" to vibration "isolation". Damping reduces the peak at the resonance but it also reduces roll-off, which means frequency above the natural frequency are amplified more. What we are trying to do is isolate, not just dampen.

Maybe for the advanced user we could add suggestions how to measure the natural frequencies of a Vibration isolation system?

hamishwillee, 2019-02-06:
@RomanBapst We can from a docs point of view refer to this as **Vibration Damping/Isolation** or **Vibration Management** to capture both cases. 

@bresch @priseborough @RomanBapst @bkueng Any comment to Roman's suggestion above? https://github.com/PX4/px4_user_guide/issues/434#issuecomment-460868033

bkueng, 2019-02-07:
@hamishwillee I agree with @RomanBapst. I think wherever I added something I used vibration isolation, and not damping.

hamishwillee, 2019-02-12:
@bkueng @RomanBapst I think you are suggesting that we globally start refering to vibration isolation rather than damping? Or are you just suggesting that *this discussion* should focus on vibration isolation?

In either case, regarding:

> "Maybe for the advanced user we could add suggestions how to measure the natural frequencies of a Vibration isolation system?"

Sounds good. Who can do this, and when? Also, what benefit does knowing these have?

priseborough, 2019-02-12:
We are asking users to isolate the board from vibration. Isolation is the accepted industry term for the objective and and involves designing a mechanical attachment with appropriate stiffness and damping properties that together with the mass/inertia of the autopilot, provides the required isolation from vibration of the airframe.

priseborough, 2019-02-12:
Damping as an objective is normally something done to the airframe itself to reduce the amplitude of specific structural resonances that can cause the autopilot mounting location to experience excessive vibration levels to a level that the isolation mount + autopilot can handle.

hamishwillee, 2019-02-15:
Thanks for the definitions Paul - helpful for me. 

I [updated](https://github.com/PX4/px4_user_guide/pull/437) the basic mounting docs to use the correct terminology (isolation).

As before, up to you guys to drive these docs. I don't know enough, so if you leave it with me, nothing will happen. 

I'm not even sure where it would "live" in the doc structure. Probably in "Advanced Configuration" as a topic *Vibration Management*, including the definitions from Paul above and Beat's [Flight Review > Vibration >solutions](https://docs.px4.io/en/log/flight_review.html#solutions)

hamishwillee, 2019-05-14:
@bkueng @priseborough @RomanBapst 

I don't know enough to do anything else useful with vibration damping docs, so if we are to do anything else, it needs to be driven by you guys. 

Can you please all respond with a three point answer:
- Vibration docs covering topic X would be useful for a significant number of users, where X is a topic
- These are sufficiently useful that it is worth putting in the effort
- I can help produce these docs in the coming Y months

There was one idea above for example: https://github.com/PX4/px4_user_guide/issues/434#issuecomment-462574938 but I have no idea if this is useful or priority. 

If we're not going to do anything because we have no good ideas, or our good ideas are not sufficient value to ever be priority, then lets close this.

bresch, 2019-05-15:
@hamishwillee  I have two interesting references about vibration analysis: 
- An Introduction to Shock & Vibration Response Spectra, Tom Irvine ([free ebook](https://info.mide.com/hubfs/eBooks/ebook-tom-irvine-shock-vibration-response-spectra.pdf))
- Structural Dynamics and Vibration in Practice - An Engineering Handbook, Douglas Thorby ([preview](https://books.google.ch/books?id=PwzDuWDc8AgC&printsec=frontcover#v=onepage&q&f=false))

Those are not cookbooks on "how to isolate your autopilot from vibrations on you hobby drone" but they are useful to gain more knowledge and understand what's going on.

hamishwillee, 2019-05-16:
Thanks very much @bresch - I think that will be useful. Added in #482

hamishwillee, 2019-06-02:
CLosing this for now. We have no further concrete actions, so there is nothing to do.

closed 2019-06-02 by hamishwillee