PX4/PX4-user_guide #439: Telemetry Radios - Need to add something on LTE

hamishwillee, 2019-02-17:
[Telemetry Radios](https://docs.px4.io/en/telemetry/) refers to Sik radio and Telemetry Wiki, but not LTE (or more generically "telephony module"?).

Can you point me to info (or people) that explain how this works with PX4? Ie 
- Are there any "protocols" for the connection
- What drivers are used
- Any specific models or versions that we know work.

Basically enough for people to be able to find something as "turnkey" AND enough info for them to find a particular module and be able to work out if it is likely to be useful.

This was triggered by http://discuss.px4.io/t/4g-lte-pixhawk-integration/9615

julianoes, 2019-02-18:
I have no idea I'm afraid. I know that my phone does LTE and 4G and that's about it :smile:.

bkueng, 2019-02-18:
I think the reason why we don't have any docs is because it has nothing to do with PX4. Essentially you can take any LTE dongle (usually via USB) and connect them to the companion. Then make sure the device works (you might need linux drivers for that) and you have connectivity.
After that it's essentially the same as any WiFi or Ethernet connection.

But yes I think adding the models we know that work makes sense (although it will also depend on the companion that you use).

hamishwillee, 2019-02-18:
Thanks @julianoes @bkueng I was envisaging that you would connect the LTE device direct to the FC and then communicate with it via LTE connected to your computer - ie a direct replacement for your telemetry radio.

However my understanding of what you're saying is that 
- LTE is used to connect your companion and other systems. 
- To the companion the dongle is just an internet connection; a ground station would connect to this over the internet. Is there a standard way for the ground station to get the dongle IP?
- Do you know the models that work - suggest we add to this page: https://docs.px4.io/en/peripherals/companion_computer_peripherals.html ?

bkueng, 2019-02-19:
> Thanks @julianoes @bkueng I was envisaging that you would connect the LTE device direct to the FC and then communicate with it via LTE connected to your computer - ie a direct replacement for your telemetry radio.

No, that would be cool.

>   LTE is used to connect your companion and other systems.

Right.

>  To the companion the dongle is just an internet connection; a ground station would connect to this over the internet. Is there a standard way for the ground station to get the dongle IP?

Correct. No standard way. Typically you can use a VPN, then you also have a secure connection.

>  Do you know the models that work - suggest we add to this page: https://docs.px4.io/en/peripherals/companion_computer_peripherals.html ?

We use the [Huawei E8372](https://consumer.huawei.com/en/mobile-broadband/e8372/) and Huawei E3372. The first one has an included WiFi, which makes development workflow a bit easier (you can keep it plugged into the companion and configure the SIM by connecting to the WiFi). The second one is a bit smaller and without WiFi (you can configure it by plugging the stick into a laptop, the WiFi is not required).

PR #443 (2019-03-04): Add LTE information
Fixes #439
closed 2019-03-04 by hamishwillee