PX4/PX4-user_guide #447: Should we use a bot to compress images on all docs

hamishwillee, 2019-02-24:
https://github.com/marketplace/imgbot

@mhkabir Thanks for link above. I'm in two minds about automating image compression. Obviously smaller files makes things better for end users. My concern is that authors will not think about image size any more - and then anyone syncing the repo ends up having to carry those big uncompressed files for local testing. 

@bkueng Thoughts? I am inclined to try this even so.

bkueng, 2019-02-25:
> @bkueng Thoughts? I am inclined to try this even so.

I agree, if it just adds commits it's not great. If it rewrites them or tells you 'hey please change your commits to use that image', it can be a nice solution.

hamishwillee, 2019-02-26:
Hi @bkueng 

It just adds commits/PR. I tried this on my local repo: https://github.com/hamishwillee/px4_user_guide/pull/3

Below is the analysis they emailed me. I don't think it is worth doing by default, but I might go through and take a small subset with significant improvements.

File | Before | After | Percent reduction
-- | -- | -- | --
/assets/hardware/pwm_esc/pwm_without_gnd.png | 19.78kb | 5.46kb | 72.40%
/assets/calibration/thermal_calibration_baro.png | 611.76kb | 272.82kb | 55.40%
/images/pixhawk_mini_pinout.png | 77.62kb | 35.27kb | 54.56%
/images/pixhawk_mini_6pin_JST_to_DF13_cable_drawing.png | 3.89kb | 1.82kb | 53.29%
/images/pixhawk_mini_6_to_6_and_4_pin_Y_cable_drawing.png | 10.42kb | 4.88kb | 53.15%
/images/pixhawk_mini_safety_switch_drawing.png | 13.98kb | 6.58kb | 52.94%
/assets/flight_log_analysis/flight_review/estimator_watchdog_good.png | 9.85kb | 4.70kb | 52.32%
/assets/hardware/sensors/px4flow/px4flow_video_distorted.png | 189.00kb | 91.37kb | 51.65%
/assets/flight_log_analysis/flight_review/logged_messages.png | 8.50kb | 4.38kb | 48.53%
/assets/site/logo_pro_small.png | 82.88kb | 43.20kb | 47.88%
/images/pixhawk_mini_compass_drawing.png | 3.87kb | 2.07kb | 46.50%
/assets/ecl/gps_glitch_-_test_ratios.png | 824.67kb | 442.67kb | 46.32%
/images/pixhawk_mini_4_pin_cable_drawing.png | 14.90kb | 8.14kb | 45.34%
/assets/ecl/gps_glitch_-_reported_receiver_accuracy.png | 254.15kb | 139.78kb | 45.00%
/assets/hardware/sensors/px4flow/test_pattern_376x240.png | 3.23kb | 1.78kb | 44.73%
/images/qgc_ekf2_enable_flow.png | 52.79kb | 29.27kb | 44.56%
/assets/camera/qgc_geotag.png | 347.20kb | 198.06kb | 42.96%
/assets/precision_land/precland-flow-diagram.png | 101.53kb | 59.38kb | 41.51%
/images/pixhawk_mini_port_main_out.png | 359.53kb | 216.62kb | 39.75%
/assets/hardware/hardware-mindracer-specs.png | 27.94kb | 16.88kb | 39.60%
/assets/hardware/hardware-nanomind110-specs.png | 28.36kb | 17.22kb | 39.27%
/assets/hardware/hardware-mindracer210-specs.png | 28.01kb | 17.04kb | 39.14%
/images/logo_pro_small.png | 4.75kb | 2.91kb | 38.78%
/assets/calibration/thermal_calibration_accel.png | 565.51kb | 361.96kb | 35.99%
/assets/ecl/gps_glitch_-_velocity_innovations.png | 436.50kb | 279.78kb | 35.90%
/assets/ecl/vibration_metrics_-_successful.png | 464.79kb | 300.42kb | 35.37%
/assets/flight_controller/crazyflie/crazyflie2_hero.png | 146.57kb | 95.47kb | 34.86%
/assets/calibration/thermal_calibration_gyro.png | 697.62kb | 455.69kb | 34.68%
/images/pixhawk_3dr_power_module.png | 257.91kb | 169.27kb | 34.37%
/images/pixhawk_3dr_power_module - Copy.png | 257.91kb | 169.27kb | 34.37%
/assets/camera/trigger_pins.png | 109.52kb | 71.95kb | 34.31%
/assets/ecl/test_ratios_-_successful.png | 459.02kb | 304.08kb | 33.75%
/images/autologic.png | 21.07kb | 13.96kb | 33.74%
/assets/flight_controller/pixhawk4/pixhawk4_receiver_sbus.png | 165.68kb | 112.06kb | 32.37%
/images/pixhawk_4_receiver_ppm.png | 134.55kb | 91.08kb | 32.31%
/images/pixhawk_mini_mounting_arrow.png | 942.67kb | 642.52kb | 31.84%
/images/vibration_aliasing.png | 274.16kb | 188.98kb | 31.07%
/assets/ecl/gps_data_loss_-_velocity_innovations.png | 130.86kb | 90.23kb | 31.05%
/images/fc_heading_mark_1.png | 111.41kb | 78.05kb | 29.94%
/assets/flight_controller/pixhawk4/pixhawk4_mounting_and_foam.png | 75.18kb | 53.07kb | 29.42%
/assets/hardware/hardware-mindpx.png | 441.20kb | 313.22kb | 29.01%
/assets/hardware/hardware-nanomind110.png | 321.47kb | 228.42kb | 28.95%
/images/speed_from_angle.png | 12.10kb | 8.73kb | 27.91%
/images/esc_calibration_step_3.png | 24.39kb | 17.76kb | 27.19%
/assets/hardware/hardware-mindpx-mounting.png | 198.10kb | 144.30kb | 27.16%
/assets/site/difficulty_easy.png | 2.14kb | 1.57kb | 27.00%
/images/qav250_auv_x2.png | 1,117.04kb | 817.31kb | 26.83%
/assets/site/difficulty_medium.png | 2.13kb | 1.57kb | 26.50%
/assets/hardware/hardware-mindpx-wiring2.png | 362.64kb | 266.88kb | 26.41%
/assets/site/difficulty_hard.png | 2.12kb | 1.57kb | 26.23%
/images/esc_calibration_step_4.png | 28.94kb | 21.66kb | 25.15%
/assets/flight_log_analysis/flightplot_0.2.16.png | 246.15kb | 185.36kb | 24.70%
/images/qgc_flight_mode_selection_qav250.png | 151.04kb | 114.08kb | 24.47%
/assets/hardware/hardware-mindracer210.png | 360.43kb | 272.26kb | 24.46%
/images/qgc_esc_calibration_power.png | 218.18kb | 164.84kb | 24.45%
/images/qgc/setup/parameters_tools_menu.png | 29.01kb | 21.92kb | 24.44%
/assets/hardware/hardware-mindracer155.png | 434.14kb | 328.41kb | 24.35%
/assets/flight_controller/pixhack_v5/pixhack_v5_connector.jpg | 119.72kb | 90.58kb | 24.34%
/images/frame_heading.png | 165.64kb | 125.35kb | 24.33%
/assets/hardware/intel_aero/intel-aero-rtf.jpg | 516.52kb | 391.54kb | 24.20%
/images/qgc_firmware_menu.png | 133.36kb | 101.87kb | 23.62%
/images/fixedwing_dtrim.png | 15.42kb | 11.84kb | 23.19%
[... 419 lines cut ...]
/images/red_parts_robocat.jpg | 159.07kb | 157.69kb | 0.87%
/images/fc_orientation_qgc_v2.png | 350.77kb | 347.92kb | 0.81%
/assets/airframes/multicopter/qav250_holybro_pixhawk4_mini/11b_escs.jpg | 225.36kb | 223.59kb | 0.79%
/assets/airframes/multicopter/qav250_holybro_pixhawk4_mini/5_power_module_on_screws.jpg | 211.55kb | 209.95kb | 0.75%
/images/auav_x2_airspeed_setup_3.png | 323.53kb | 321.18kb | 0.72%
/assets/airframes/multicopter/qav250_holybro_pixhawk4_mini/10_aluminium_standoffs.jpg | 225.71kb | 224.12kb | 0.71%
[... cut ...]
/images/airframes/multicopter/qav-r/qav-r.png | 946.57kb | 944.36kb | 0.23%
/assets/hardware/sensors/px4flow/flow_mounting_iris_2.png | 971.62kb | 969.58kb | 0.21%
/assets/hardware/sensors/px4flow/flow_mounting_iris.png | 919.65kb | 917.73kb | 0.21%
/assets/hardware/sensors/px4flow/flow_focus_book.png | 831.27kb | 829.91kb | 0.16%
/images/red_robo_naked_side.jpg | 104.88kb | 104.74kb | 0.14%
/assets/airframes/multicopter/qav250_holybro_pixhawk4_mini/15b_attach_gps.jpg | 194.53kb | 194.39kb | 0.07%
  |   |   |  
Total : | 104,396.25kb | 92,980.89kb | 10.93%

hamishwillee, 2019-02-26:
And FYI, I did a removal/reduction of some of the larger files in https://github.com/PX4/px4_user_guide/pull/449 - these were mostly the very large ones that significantly affect page load, or which are no longer used at all.

bkueng, 2019-02-28:
> Below is the analysis they emailed me. I don't think it is worth doing by default, but I might go through and take a small subset with significant improvements.

Looking at the list, I agree with that.

hamishwillee, 2025-01-29:
Just FYI, on new PRs I do a manual reduction in file size. That means generally we benefit from reduction. It isn't automatic but it doesn't eat much time when doing a review and IMO it is "good enough".

Closing as stale. We can of course revisit if there is a strong push at some point.

closed 2025-01-29 by hamishwillee