PX4/PX4-user_guide #476: How to install docs locally?

fnoop, 2019-05-03:
README.md says:
> Everything you need to install and build Gitbook locally is explained in its toolchain documentation.

However the link to gitbook goes to v2 changelist which says there is no more local/offline install.  If I want to take a copy of the docs offline for field reference, how can I do this?  Are the docs now completely dependent on proprietary closed 3rd party?

hamishwillee, 2019-05-06:
I've fixed up our docs to point to the original installation instructions on github. 

So no, our docs still use the legacy gitbook system. At some point this might have to change but right now the system does what we need.

hamishwillee, 2019-05-06:
Fixed  by https://github.com/PX4/Devguide/commit/dcf121ef5034c066bf284e0f561234639364116e

hamishwillee, 2019-05-06:
PS those instructions are here https://github.com/GitbookIO/gitbook/blob/master/docs/setup.md

closed 2019-05-06 by hamishwillee

fnoop, 2019-05-08:
Works great, thanks :)