PX4/PX4-user_guide #488: Lightware SFxx Lidar - What do and should we support

hamishwillee, 2019-05-27:
[Lightware SFxx Lidar ](https://docs.px4.io/master/en/sensor/sfxx_lidar.html) lists a few rangefinders.
1. Lightware only LISTS these ones -https://lightware.co.za/pages/product-overview 
   - should we be moving the ones that they aren't supporting anymore into separate list - ie SF02, 10A, 10B, 10C
   - Do we support the other models listed here?
1. We say LW20 , but spec refers to LW20/B, LW20C, LW20/SER/X - which do we actually support? All?

@bkueng If you don't know, who would you recommend we ask?

bkueng, 2019-05-27:
@DanielePettenuzzo might know

hamishwillee, 2019-05-28:
Thanks @bkueng - one question still there for you "should we be moving the ones that they aren't supporting anymore into separate list - ie SF02, 10A, 10B, 10C". Ie we support them, but I don't want to recommend them.

@DanielePettenuzzo any ideas on what versions of LW20 we support? Is there a good way of us making sure we keep in step with lightware?

bkueng, 2019-05-28:
> Thanks @bkueng - one question still there for you "should we be moving the ones that lightware aren't supporting anymore into separate list - ie SF02, 10A, 10B, 10C". Ie we support them, but I don't want to recommend them. If we are going to keep them front and center we need to have clear to buy links, and I'm not sure that there will be options to do so if manufacturer isn't pushing them

Yes that makes sense.

hamishwillee, 2019-05-28:
Thanks @bkueng - discontinued now separated out.

@DanielePettenuzzo Please let me know:
- Do we support the other models in https://lightware.co.za/pages/product-overview 
- Do we have a roadmap for supporting the others. 
- We say we support LW20 , but spec refers to LW20/B, LW20C, LW20/SER/X - which do we actually support? All?

PR #499 (2019-05-28): SFxxx rangefinder: separate out discontinued
A bunch of lightware rangefinders are no longer available from manufacturer. This identifies active vs discontinued and does some tidy of presentation. Partially fixes #488