PX4/PX4-user_guide #490: QuadPlane Configuration - VT_TRANS_THR does not exist

hamishwillee, 2019-05-27:
[QuadPlane Configuration > Transition throttle](https://docs.px4.io/master/en/config_vtol/vtol_quad_configuration.html#transition-throttle)

This links to VT_TRANS_THR which does not exist. Did it get replaced or disappear?

@RomanBapst Is this a question for you?

xdwgood, 2019-12-09:
@hamishwillee 

`VT_F_TRANS_THR`---> Target throttle value for the transition to fixed wing flight.
`VT_B_TRANS_THR`---> Target throttle value for the transition to hover(Multi-rotor) flight

But I didn't find the setting of transmission throttle in tailsitter type

hamishwillee, 2019-12-11:
Thanks. We need @RomanBapst to comment - it isn't just a matter of replacing the existing parameter in docs.

xdwgood, 2019-12-11:
Yes, I already know that in tailsitter drones, the thrust when switching to fixed-wing mode is generated by multi-rotor altitude control
Instead of a parameter, we do need to clarify some instructions :smiley:

hamishwillee, 2025-01-29:
The current docs at this location include a [working link to the parameter](https://docs.px4.io/main/en/config_vtol/vtol_quad_configuration.html#transition-throttle). So I presume this has been updated.

Closing.

closed 2025-01-29 by hamishwillee