PX4/PX4-user_guide #495: Doc Bug: Wing Wing Z84 (Pixracer) - Where find ePower-Pro-12A?

hamishwillee, 2019-05-27:
[Wing Wing Z84 (Pixracer) · PX4 User Guide](https://docs.px4.io/master/en/frames_plane/wing_wing_z84.html)

Broken link to http://www.eflight.ch/pi/ePower-Pro-12A.html failed - any ideas where it might be sourced?

bkueng, 2019-05-28:
The only place I find it is on hobbyking: https://hobbyking.com/de_de/wing-wing-z-84-epo-845mm-kit.html?___store=de_de

hamishwillee, 2019-05-28:
Hi @bkueng - that links is for the wing wing, not the ePower pro esc. Search doesn't surface it.

bkueng, 2019-05-29:
Ha, I should read properly.
For that it doesn't really matter, pretty much any small, >=12A ESC will do. For example this: https://www.brack.ch/lumenier-regler-30a-blhelis-esc-opto-2-4s-589152?query=esc&redirected=1

hamishwillee, 2019-05-30:
Thanks very much! Fixed.

closed 2019-05-30 by hamishwillee