PX4/PX4-user_guide #514: Not clear OFFBOARD mode

tuloski, 2019-07-08:
Here https://docs.px4.io/en/flight_modes/offboard.html it's written that in OFFBOARD mode the vehicle position, velocity, or thrust (SET_POSITION_TARGET_LOCAL_NED) can be controlled.

What does the OR stands for? What if I send a SET_POSITION_TARGET_LOCAL_NED with position AND velocity (AND thrust)? Or should I only pick one to send (position or velocity or thrust)?

Can we have more info on how the setpoint is created and if it is useful or useless to send both position and velocity.

hamishwillee, 2019-07-09:
@julianoes Can you comment on whether you can control position, velocity and thrust at the same time in offboard mode  with [SET_POSITION_TARGET_LOCAL_NED](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_LOCAL_NED)? Or can you only control one.

Looking at the code you can *set* them all, but I can't work out whether the system will try an act on a setpoint that specifies position, velocity and thrust (and I imagine behaviour would be quite unpredictable).

tuloski, 2019-07-09:
> Looking at the code you can _set_ them all at the same time and these are published to a UORB topic, but I can't work out whether the system will try an act on a setpoint that specifies position, velocity and thrust (and I imagine behaviour would be quite unpredictable).

Yeah same. I'll try to look better at the code to understand this.

julianoes, 2019-07-09:
> Yeah same. I'll try to look better at the code to understand this.

No idea, I'd have to do the same, sorry. Maybe @MaEtUgR or @Stifael know.

hamishwillee, 2019-08-01:
Tried to get an answer here: https://px4.slack.com/archives/C3B9NSHRQ/p1564638280024600

Stifael, 2019-08-01:
SET_POSITION_TARGET_LOCAL_NED will eventually be handled in FlightTaskOffboard and currently it supports the following combination of setpoints: https://github.com/PX4/Firmware/blob/master/src/lib/FlightTasks/tasks/Offboard/FlightTaskOffboard.cpp#L166-L170

Important to know it that there is no support for acceleration (independent of FlightTask), but rather acceleration is mapped directly to thrust-setpoint.

Stifael, 2019-08-01:
However, I honestly would first test if the current mavlink_interface actually is doing the right mapping (https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L810-L819)

A long time ago I started this branch (https://github.com/Stifael/Firmware/commit/afab1f6c3e277f3f8d682dfcf57f1715f6cb8795) together with this mavsdk changes (https://github.com/mavlink/MAVSDK/pull/434) which allow for additional combination

MaEtUgR, 2019-08-02:
> @MaEtUgR is working on acceleration support (not sure about the current state).

It's part of #12072. I wanted to be finished a long time ago but found small bugs and was always interrupted while fixing them with "more important" things. My next goal is to really finish this up now.

hamishwillee, 2019-08-05:
Thanks very much @Stifael and @MaEtUgR . I have created #545 to fix this - reflecting exactly what it says is supported in code. Note that I haven't actually "checked" the assignment - bit out of scope for a doc update.

PR #545 (2019-08-06): Clarify which setpoints supported
Fixes #514 

@Stifael @MaEtUgR Would appreciate your review. Comments inline.
closed 2019-08-06 by hamishwillee