PX4/PX4-user_guide #561: Document mappydot distance sensor

hamishwillee, 2019-08-20:
https://github.com/PX4/Firmware/pull/12147

@mhkabir You did the work on this right? Can you create some basic docs, or talk me though it in bullet form here. Not much point adding a driver for anything without how to use docs.

mcsauder, 2019-08-20:
@hamishwillee , I wrote this driver, i am glad to help with the docs. Let me get a start on it and ask for your help to polish things up!

hamishwillee, 2019-08-20:
Thanks @mcsauder - much appreciated!

Just to give you a starting point
1. Good to have some docs here: https://dev.px4.io/master/en/middleware/modules_driver_distance_sensor.html but you can do those last
2. Distance sensors typically have a very high level introduction here: http://docs.px4.io/master/en/sensor/rangefinders.html - mostly this is just a very high level introduction to the sensor and link to dedicated page. By introduction I mean "this is how the marketing for the sensor self describes + some info on distances and key selling point"
3. Docs structure for individual page can be inferred from examples - they are all pretty much the same. Basically though you want people to know what it is (range, key characteristics),  be able to wire the specific hardware, any specific configuration, links to any general configuration.
If there is anything special about the driver setup add that too - e.g. this one (only) can be used for collision prevention, so I added that info: http://docs.px4.io/master/en/sensor/cm8jl65_ir_distance_sensor.html#parameter-setup

Some sensors can be used on I2C, others UAVCAN, others Serial. So we try capture what works with what.

hamishwillee, 2025-01-29:
@mcsauder I'm closing/tidying up issues that are very stale and unlikely to be addressed, of which this is one.

The current state is that mappydot has a module page and nothing much else: https://docs.px4.io/main/en/modules/modules_driver_distance_sensor.html#mappydot.
I have created a link at the top of the rangefinder page that links to the modules, so perhaps people can discover these.

If you would like to create a section in the rangefinders docs for this it would be great. 

Closing for now.

closed 2025-01-29 by hamishwillee