PX4/PX4-user_guide #565: Return Mode ·- RTL_TYPE needs documentation

hamishwillee, 2019-08-23:
Need to update docs with info about [RTL_TYPE](http://docs.px4.io/master/en/advanced_config/parameter_reference.html#RTL_TYPE), and specifically the fact that it is enabled by default on FW - https://github.com/PX4/Firmware/pull/12770

I imagine we need to update http://docs.px4.io/master/en/flight_modes/return.html and maybe http://docs.px4.io/master/en/flying/fixed_wing_landing.html

hamishwillee, 2019-08-23:
@RomanBapst Saw your note in https://github.com/PX4/Firmware/pull/12770#issuecomment-523772935 thanks. A couple of questions:
1. Does mission mode landing just mean a mission with a waypoint set using [MAV_CMD_NAV_LAND](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LAND)?
2. Does this type apply to MC as well - or just FW and VTOL? 
3. Is it default for VTOL and MC too?
4. Do all the RTL_TYPES work? Ie will it reverse the mission if specified
5. What is the path to the point where it starts a landing? Ie the RTL_TYPE says direct, but what about obeying the altitude rules of normal RTL? E.g. is all this "cone-like" behaviour ignored? http://docs.px4.io/master/en/flight_modes/return.html#multi-copter-mc ?

Antiheavy, 2019-08-23:
If it's okay I'll take a stab at some answers:

> Does mission mode landing just mean a mission with a waypoint set using [MAV_CMD_NAV_LAND](https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LAND)?

In this context several things are required: 1) a DO_LAND_START https://mavlink.io/en/messages/common.html#MAV_CMD_DO_LAND_START, 2) at least one position waypoint, and 3) a Land waypoint https://mavlink.io/en/messages/common.html#MAV_CMD_NAV_LAND.  These items are required at a minimum to define a landing pattern that can be used for RTL_TYPE 1 and 2.

> Does this type apply to MC as well - or just FW and VTOL?

I'm not sure about MC, but in my opinion it should be an option for MC too.  No reason to prevent a MC from using an appropriately designed mission landing pattern.

> Is it default for VTOL and MC too?

Currently (recently) default for VTOL and FW only.

> Do all the RTL_TYPES work? Ie will it reverse the mission if specified

Last I tested RTL_TYPE 2 it worked, but it's been a while since I've tried.

>     5\. What is the path to the point where it starts a landing? Ie the RTL_TYPE says direct, but what about obeying the altitude rules of normal RTL? E.g. is all this "cone-like" behaviour ignored? http://docs.px4.io/master/en/flight_modes/return.html#multi-copter-mc ?

I'm not sure about MC, but with FW (And I assume VTOL in FW mode) the first step is the min_Return_altitude climb if starting below it, then a direct flight to the first waypoint in the pattern.  I think "Cone" only applies to MC? (i hope?)

hamishwillee, 2019-08-29:
Thanks @Antiheavy . I've been dragged away to other tasks. This is very helpful though, and I will get back to it soon

hamishwillee, 2019-09-02:
@Antiheavy @RomanBapst So historically for a VTOL, on return it would ALWAYS transition back to MC mode and land at home.

Can you confirm what happens now?
1. IF there is no mission or there is a mission but no mission land sequence, will it just switch to MC and land at home?
2. If there is a mission with a mission land sequence, does this mean that it lands as an FW, or is that something that is defined in the mission? (and if so how)?

hamishwillee, 2019-09-02:
@Antiheavy , @RomanBapst RTL)TYPE = 2 is 

> 2: Return to a planned mission landing, if available, using the mission path, else return to home via the reverse mission path

How does this work (a) if you're in a mission, (b) if you're not.

If you're in a mission it reads like "if there is a landing sequence just continue to it, but if not, reverse your mission back to the beginning"
- Is that correct?
- When you get back to the beginning of the mission, what do you do? Just circle/follow what the RTL_LAND_DELAY parameter says? 

If you're not in a mission what happens? Presumably you join the mission path - but where - do you just choose the closest waypoint at RTL time to you and join the mission. Or do you choose the closest mission forward if there is a landing point and the closest back to start of mission if there isn't?

hamishwillee, 2019-09-02:
And last of all, does it obey the "cone" params for MC (http://docs.px4.io/master/en/flight_modes/return.html#multi-copter-mc) in FW and VTOL modes?

Antiheavy, 2019-09-03:
> If you're in a mission it reads like "if there is a landing sequence just continue to it, but if not, reverse your mission back to the beginning"
> 
>     * Is that correct?

Last I tested in SITL (long time ago) this is basically correct.  The RTL=2 will follow the mission waypoints forward to the planned landing pattern, however, it will skip over loiters that have time delay or other elements like that and just keep going right through all the waypoints to the end with the landing.  If no landing pattern is there, the RTL=2 will reverse the mission, skipping over any time delays, and then once at the beginning of the mission it will fly to Home and loiter there, doing whatever normal RTL at Home things are programmed via the parameters.

> If you're not in a mission what happens? Presumably you join the mission path - but where - do you just choose the closest waypoint at RTL time to you and join the mission

Yes, the vehicle goes to the nearest waypoint in space, doesn't matter what order.

hamishwillee, 2019-09-05:
Thanks for all your help @Antiheavy - you can see the fruits of this in https://github.com/PX4/px4_user_guide/pull/571

closed 2019-09-16 by hamishwillee