PX4/PX4-user_guide #577: Doc Bug: BetaFPV Beta75X 2S Brushless Whoop · PX4 v1.9.0 User Guide

looninho, 2019-09-21:
No Air Frame: BetaFPV Beta75X 2S Brushless Whoop in QGC

Bug Page: [BetaFPV Beta75X 2S Brushless Whoop · PX4 v1.9.0 User Guide](https://docs.px4.io/v1.9.0/en/complete_vehicles/betafpv_beta75x.html)

hamishwillee, 2019-09-23:
Are you sure? I can see it in the [Airframe reference](https://docs.px4.io/v1.9.0/en/airframes/airframe_reference.html#copter_quadrotor_h_betafpv_beta75x_2s_brushless_whoop). Did you look in Quadrotor H?
[image]
I tested against master branch, but you should still see this in 1.9 because QGC uses the same source as that airframe reference doc.

looninho, 2019-09-23:
Yes, thank you, found it in H frame. I was looking in X frame!

I found it more stable than betaflight but I have to tune the PID in yaw axis because it woobles when descending and rotates when I lower the throttle.

I'm searching to make OSD working. If a good man can implement the flip after crash mode...

best regards

hamishwillee, 2019-09-24:
Thanks for confirming. If you have general questions, best place to start is https://discuss.px4.io/

closed 2019-09-24 by hamishwillee

bkueng, 2019-09-24:
> I found it more stable than betaflight but I have to tune the PID in yaw axis because it woobles when descending and rotates when I lower the throttle.

Let me know when you have better tuning gains. FYI master has switched to 1kHz rate loop and there's a PR for DShot (which I already tested on the whoop) - both improve the flight performance.