PX4/PX4-user_guide #604: Camera Capture Driver docs

hamishwillee, 2019-11-10:
Hi @DanielePettenuzzo 

You got a plan to do these docs as you stated in the links below?
https://github.com/PX4/Firmware/pull/10295
https://github.com/PX4/Firmware/pull/11395

Cheers
H

Jaeyoung-Lim, 2021-12-08:
I am struggling to get the input capture working with a pixhawk4 and it seems like the doc is not fully capturing https://github.com/PX4/PX4-Autopilot/pull/11395

hamishwillee, 2025-01-29:
The docs have had some updates since this was posted - currently are in https://docs.px4.io/main/en/camera/fc_connected_camera.html#camera-capture-configuration

Happy to take feedback on the current docs, or specific advice for how they might be improved. But for now, consider taking this further to be low value for potentially a lot of work

Closing.

closed 2025-01-29 by hamishwillee