PX4/PX4-user_guide #685: Doc Bug: Hex Cube (Pixhawk 2) (FMUv3) · PX4 User Guide

vivienJ, 2020-03-18:
DESCRIBE PROBLEM WITH DOCS HERE

Bug Page: [Hex Cube (Pixhawk 2) (FMUv3) · PX4 User Guide](https://docs.px4.io/master/en/flight_controller/pixhawk-2.html)

Hello, there is a mistake in the pdf link here "(The datasheet from Hex manufacturing can be found here." (http://www.hex.aero/wp-content/uploads/2016/07/DRS_Pixhawk-2-17th-march-2016.pdf)
Page 15 in the table "Power 6 pos" the BATT_VOLTAGE_SENS_PROT and BATT_CURRENT are switched on the pixhawk cube black. The pinout must be 5V | 5V | current | voltage | GND | GND.

You can find here a pdf without this mistake : https://discuss.ardupilot.org/t/pixhawk-2-1-pinout-doc-need-to-be-on-the-pixhawk-2-1-page/23604

hamishwillee, 2020-03-19:
Can you please tell me the exact PDF, at that link I see the same PDF - http://www.hex.aero/wp-content/uploads/2016/07/DRS_Pixhawk-2-17th-march-2016.pdf

vivienJ, 2020-03-19:
Yes this is the PDF link on the PX4 documentation at the line "The datasheet from Hex manufacturing can be found here". It is the one who has an  error page 15.

hamishwillee, 2020-03-20:
I mean that I can't work out which PDF you are talking about that has the fix.

hamishwillee, 2020-03-20:
Hi @proficnc Do you think you could confirm the error in http://www.hex.aero/wp-content/uploads/2016/07/DRS_Pixhawk-2-17th-march-2016.pdf and point to an updated version?

hamishwillee, 2020-04-03:
Thanks @vivienJ and @proficnc  Fixed.

commit cf2a49e2 (2020-04-03): Cube docs: Fixes link from old PDF spec to doc site

Fixes #685
closed 2020-04-03 by hamishwillee