PX4/PX4-user_guide #699: USB Port numbers and version are very confusing in PX4 Vision Kit docs

bys1123, 2020-03-26:
[image]
[image]
@Jaeyoung-Lim

Jaeyoung-Lim, 2020-03-26:
@bys1123 Thanks for the feedback. Would it be possible to open a PR that fixes the confustion? thanks

bys1123, 2020-03-26:
@Jaeyoung-Lim No problem, I'll do that tomorrow.

hamishwillee, 2020-03-26:
@Jaeyoung-Lim @bys1123 I made the fixes in https://github.com/PX4/px4_user_guide/commit/ed17b1e1000e8e426f69868c41f3f1d9129188f6

BUT this is not the solution. The solution is for the casing to be given better labeling - i.e. USB1 > USB-1, USB2 > USB-2, UART1 to UART-1
@jinger26 Can you request this?

closed 2020-03-26 by hamishwillee

bys1123, 2020-03-27:
Why? I don't see many products labeled with "-" on casing.

bys1123, 2020-03-27:
It makes you can only see the "-" when you first get the product.
Same - element - are - easy - to - be - connected - in - your - mind

hamishwillee, 2020-03-27:
@bys1123 Because every time someone sees USB2 or USB3 on the case they think it is a USB2 or USB3 port.

@jinger26 Perhaps a better way might be to use a #. So `USB#1`, `USB#2` - since # actually means "number".

bys1123, 2020-03-27:
I would suggest `USB①` `USB②` `USB③`

hamishwillee, 2020-03-27:
Sure. I don't care as long as it is clearly a number and not a USB version :-)

bys1123, 2020-03-27:
Just checked some designs, is the numerical layout really necessary
[image]

jamming, 2020-04-01:
It is very difficult to clearly printing some symbols(such as '#' or '②' )on the 3D print case. So in the next batch of products, all USB 2 version ports will be identified as 'USB2.0'.  All USB 3 version ports will be identified as 'USB3.0'.  USB port number will no longer be printed on the case.

bys1123, 2020-04-01:
[image]
@jamming Will this works?

hamishwillee, 2020-04-01:
@jamming That's disappointing, but pragmatic and workable.

jamming, 2020-04-02:
> [image]
> @jamming Will this works?
 @bys1123 
①② can't clearly printing on the 3D print case