PX4/PX4-user_guide #725: Airframe config for tricopter configuration not correct

Jaeyoung-Lim, 2020-04-20:
The airframe config for the tricopter airframe seems to have incorrect prop directions. 

As in: https://docs.px4.io/master/en/airframes/airframe_reference.html#tricopter-y

sfuhrer, 2020-04-20:
The direction of the propellers isn't taken into account by the software, we should add this comment there. And also change the picture to have the front propellers turning in opposite directions, as this is how you do it for all systems I know.

hamishwillee, 2020-04-21:
This page is built from the airframe definititon - so any comments have to be added there.

I actually don't know where/how these images are created, but they need to also be changed in the Devguide and in QGC. 

@bkueng Do you know where the images come from/what tools are used to create them?

bkueng, 2020-04-21:
I used inkscape in the past, but don't know where they come from.

hamishwillee, 2020-04-21:
@julianoes Any ideas on where we get images for the airframes - or who might know?

julianoes, 2020-04-21:
> @julianoes Any ideas on where we get images for the airframes - or who might know?

No idea.

DanAbara, 2021-07-11:
Hi @hamishwillee , I will like to help with this issue. Please assign to me. I did some checks and the images used in the airframe configs are in https://github.com/PX4/PX4-user_guide/tree/master/assets/airframes/types. 

As per @sfuhrer's comments, I've modified the images so that the front propellers turn in opposite directions. 

                          
[image] [image]

              Tri Y-                           Tri Y+

Before I create a PR, just to confirm, should we have this setup (front propellers in opp. directions) as an alternative to the existing config ?, since either of them should work. Or should I just replace the existing images with this one. Thank you.

Jaeyoung-Lim, 2021-07-11:
@DanAbara Shouldn't the colors indicate the turning directions of the prop?

DanAbara, 2021-07-11:
> @DanAbara Shouldn't the colors indicate the turning directions of the prop?

You're right. I've modified them.

[image] [image]

@Jaeyoung-Lim so should I add these alternative setups or just replace the existing ones?

hamishwillee, 2021-07-11:
I'm back on Wednesday and will think a bit more on this. My initial reaction though is that this may well be a new config. The images would also need to go through to QGC.

DanAbara, 2021-07-12:
Alright @hamishwillee, just point me in the right direction when you're back and i'm happy to make the necessary changes.

hamishwillee, 2021-07-14:
@DanAbara Awesome. As you already worked out, all you need to do is create a PR that replaces YMinus.svg and YPlus.svg with your (same name) versions in https://github.com/PX4/PX4-user_guide/tree/master/assets/airframes/types
I have assigned to you.

As a separate task, it would be good to also update the images here: https://github.com/mavlink/qgroundcontrol/tree/master/src/AutoPilotPlugins/Common/Images - though these appear not to include arrow directions

DanAbara, 2021-07-14:
@hamishwillee, PR created. I've updated the images in QGC as well without the arrows and created a PR for that as well as well.

hamishwillee, 2021-07-15:
Thanks. I've reviewed the docs PR. Up to QGC folks to review the other. If nothing happens in a week or so on that ping me.

PR #1403 (2021-07-16): docs: tri airframe setup. 
### Issue
The docs for airframe config for the tricopter airframe had all propellers rotating in same direction.

### Solution
- Added a comment that the software does not take the propeller directions into account.
- Updated images in airframe config. so that front propellers turn in opposite directions.

Fixes #725
closed 2021-07-16 by hamishwillee