PX4/PX4-user_guide #740: Doc Bug: VTOL w/o Airspeed Sensor · PX4 User Guide

George-avy, 2020-05-20:
The page states:

> To tell the flight controller it will fly without an airspeed sensor you need to set the airspeed mode (FW_ARSP_MODE) to 'declare airspeed invalid' **(2)**.

whereas the correct value for FW_ARSP_MODE for disabled airspeed is 1, as per https://docs.px4.io/master/en/advanced_config/parameter_reference.html#FW_ARSP_MODE

Bug Page: [VTOL w/o Airspeed Sensor · PX4 User Guide](https://docs.px4.io/master/en/config_vtol/vtol_without_airspeed_sensor.html)

George-avy, 2020-05-20:
Same a bit below on the page: 

> QUICK OVERVIEW OF RELEVANT PARAMETERS

    FW_ARSP_MODE: Declare invalid (2)

hamishwillee, 2020-05-21:
Thanks @George-avy - much appreciated.
Fixed in master, v1.11 and v1.10 docs brances.

commit 75d2dd9e (2020-05-21): vtol without airspeed sensr

Fixes #740
closed 2020-05-21 by hamishwillee