PX4/PX4-user_guide #753: Doc Bug: Sensor Thermal Compensation · PX4 User Guide

bigbellmercy, 2020-06-15:
LOCATION:
In the section 'Offboard Calibration Procedure', there is no. 12 step as below:
"12. Power the board and perform a normal accelerometer sensor calibration using QGroundControl. It is important that this step is performed when board is within the calibration temperature range. The board must be repowered after this step before flying as the sudden offset changes can upset the navigation estimator and some parameters are not loaded by the algorithms that use them until the next startup."

PROBLEM:
But, after the accelerometer calibration is done, the parameter values for accelerometer calibration are not saved and maintain default values. So the step is meaningless actually. This problem can be reproduced in PX4 v1.10.2, v1.9.0, v1.7.2 and v1.6.5.

CONCLUSION:
Therefore, it needs to be checked whether the accelerometer calibration step is correct or not. And then the document can be clarified more.

Bug Page: [Sensor Thermal Compensation · PX4 User Guide](https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html)

hamishwillee, 2020-06-15:
@bigbellmercy When you calibrate the accelerometer (normally, not as part of this procedure) it should save the value of your calibration in parameters. Otherwise there is no point in ever calibrating the accelerometer. Are you saying that when you calibrate your hardware the values are never saved? 

If so, on what hardware?

bigbellmercy, 2020-06-16:
Yes, it is. The values are never saved in the procedure with Pixhawk2.1. (But, in normal calibration the values are saved well)

hamishwillee, 2020-06-16:
In step 11 you're resetting the system back to "normal". So at this point you **are** doing a normal calibration. It is weird if calibration works before this setup, but not after. 

If you reboot after step 11 and then reboot again would calibration work (or to put it another way, how do you get it back to normal in order to do a normal calibration?)

hamishwillee, 2020-06-16:
@MaEtUgR In step 12 of [offboard thermal calibration](https://docs.px4.io/master/en/advanced_config/sensor_thermal_calibration.html#offboard_calibration) there is a final accelerometer calibration that appears to be ignored on Cube. 
Accelerometer calibration outside of this process appears to work. 

Advice please [I've asked @bigbellmercy for a little more information above]

bigbellmercy, 2020-06-16:
> In step 11 you're resetting the system back to "normal". So at this point you **are** doing a normal calibration. It is weird if calibration works before this setup, but not after.
>
> If you reboot [...]

If the parameters TC_A_ENABLE, TC_B_ENABLE and TC_G_ENABLE are reset to zero, normal calibration is possible. 

But those parameters must be kept to 1 to utilize the result of thermal calibration I think. When the parameters are reset to zero all the TC_ parameters are disappeared in QGC.

bigbellmercy, 2020-06-16:
Another information:

I found now that there is a possibility that it may not be a problem. After thermal calibration, those normal calibration parameters may become the default values (1 or 0) because the thermal calibration compensated completely all the offsets of sensors and there are no offset remained for compensation. So users may think wrong that the sensor calibration parameter values are not saved.

If it is, such a guiding comment could be added to the text to protect such a confusion of users.

One more information:

I did 'onboard' thermal calibration (this is not 'offboard' calibration that is dealt in this issue). But the symptom was the same. After done it, the final sensor calibration (accel etc.) step was done as the procedure. But those calibration parameter values such as CAL_ were the same of the default values (1 or 0) as like the 'onboard' thermal calibration case. The same symptom seems support my guess for the confusion.

hamishwillee, 2020-06-16:
Thanks. I can't really comment until we have feedback from @MaEtUgR  (or others) on expected behaviour.

MaEtUgR, 2020-06-17:
> After thermal calibration, those normal calibration parameters may become the default values (1 or 0) because the thermal calibration compensated completely all the offsets of sensors and there are no offset remained for compensation.

Exactly. When the board is thermal calibrated the normal calibration parameters `CAL_` are not used anymore also not for the normal calibration. Instead the normal calibration incorporates its result into the thermal calibration parameters `TC_` that are in use.

PR #820 (2020-10-21): Add note about thermal vs original calib. params
@MaEtUgR I think this fixes #753 .

Can you also confirm we still don't support thermal calibration of the magnetometer?
closed 2020-10-21 by hamishwillee